OrcaSlicer/OrcaSlicer #4077: first_flush_volume and second_flush_volume values taken from Prime volume instead of Flushing volumes

evelyndooley, 2024-02-13:
### OrcaSlicer Version

1.9.0

### OS version

Windows 11

### Printer

Voron 2.4 with Trad Rack MMU

### How to reproduce

1. Enable Prime tower in process.
2. In any custom G-Code section, utilize placeholder. Example: 
```
G1 E{first_flush_volume}
G1 E{second_flush_volume}
```

### Actual results

For a Prime volume of 15 mm^3:
```
G1 E3.11814
G1 E3.11814
```

### Expected results

For flushing volumes of 70 and 150 for first and second respectively:
```
G1 E70
G1 E150
```

### Project file & Debug log uploads

[Chillet4.zip](https://github.com/SoftFever/OrcaSlicer/files/14269278/Chillet4.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

barsamgr, 2024-02-17:
You meed to activate "Purge in prime tower" of the printer settings for them to work.

[image]

evelyndooley, 2024-02-17:
Thank you, I missed that setting. It is working as expected now.

SoftFever, 2024-02-29:
Looks like the problem is solved. 
Closing

closed 2024-02-29 by SoftFever