OrcaSlicer/OrcaSlicer #4078: Alert the user if "Make overhangs printable" is active while slicing

RenanAz, 2024-02-13:
**Is your feature request related to a problem? Please describe.**
Printing something after unintentionally leaving "Make overhangs printable" left over from another print can create a lot of issues because ther are important changes to the geometry. 
Example is this week i lost a build plate full of gridfinity multi-bins because i changed to a profile that had that option active and it just printed solid without the grid valleys. They wont fit the grid now.
If we had an alert after slicing (like we have with aggressive overhangs) it would allow us to choose and avoid wasting time and material.

**Which printers will be beneficial to this feature**
Any / all printers

**Describe the solution you'd like**
Add an alert shown while slicing (like we have when file has aggressive overhangs) telling the user that they are slicing with "Make overhangs printable" enabled. 

**Describe alternatives you've considered**
Make this option not persist when saving the profile, but i think this is not a great alternative.

**Additional context**
Add some alert here mentioning "Make overhangs printable is enabled for this print - please be aware of the geometry changes applied"
[image]

NeverUsedID, 2024-02-14:
Would like customizeable alert popup's. Like 

- ironing enabled
- Make overhangs printable enabled

Are You sure you want to proceed?

SoftFever, 2024-02-22:
Since this option will change model's geometry, I think a popup dialog is more preferable.
The change is in main branch now.
[image]

commit 558510c5 (2024-02-22): Show a warning dlg when enabling `Make overhangs printable` option
Fixes #4078
closed 2024-02-22 by SoftFever