OrcaSlicer/OrcaSlicer #4079: webkitWebViewBaseEnterAcceleratedCompositingMode Segmentation fault

mabl, 2024-02-13:
### OrcaSlicer Version

1.9.0

### OS version

Nixos Unstable Package

### Additional system information

I run from stable NixOS, using an unstable package via `nix-shell -I nixpkgs=~/.nix-defexpr/channels/nixos-unstable/ -p orca-slicer`. I am using KDE with fractional scaling and usually apply the `WEBKIT_DISABLE_COMPOSITING_MODE=1 ` workaround from #2703. Unfortunately, this new crash appears with and without that flag.

### Printer

Voron 2.4

### How to reproduce

```
nix-channel --add https://nixos.org/channels/nixos-unstable nixos-unstable
nix-channel --update
nix-shell -I nixpkgs=~/.nix-defexpr/channels/nixos-unstable/ -p orca-slicer
orca-slicer
```

### Actual results

Crashes, shortly after showing initial window.

```
Thread 1 "orcaslicer_main" received signal SIGSEGV, Segmentation fault.
0x00007fffe9932e52 in webkitWebViewBaseEnterAcceleratedCompositingMode(_WebKitWebViewBase*, WebKit::LayerTreeContext const&) ()
   from /nix/store/pjprkpwc95bxqqda4n3wrxy4w23l2fzi-webkitgtk-2.42.5+abi=4.0/lib/libwebkit2gtk-4.0.so.37
(gdb) bt
#0  0x00007fffe9932e52 in webkitWebViewBaseEnterAcceleratedCompositingMode(_WebKitWebViewBase*, WebKit::LayerTreeContext const&) ()
   from /nix/store/pjprkpwc95bxqqda4n3wrxy4w23l2fzi-webkitgtk-2.42.5+abi=4.0/lib/libwebkit2gtk-4.0.so.37
#1  0x00007fffe93eb74b in WebKit::DrawingAreaProxy::didReceiveMessage(IPC::Connection&, IPC::Decoder&) ()
   from /nix/store/pjprkpwc95bxqqda4n3wrxy4w23l2fzi-webkitgtk-2.42.5+abi=4.0/lib/libwebkit2gtk-4.0.so.37
#2  0x00007fffe97154ea in IPC::MessageReceiverMap::dispatchMessage(IPC::Connection&, IPC::Decoder&) ()
   from /nix/store/pjprkpwc95bxqqda4n3wrxy4w23l2fzi-webkitgtk-2.42.5+abi=4.0/lib/libwebkit2gtk-4.0.so.37
#3  0x00007fffe983c4af in non-virtual thunk to WebKit::WebProcessProxy::didReceiveMessage(IPC::Connection&, IPC::Decoder&) ()
   from /nix/store/pjprkpwc95bxqqda4n3wrxy4w23l2fzi-webkitgtk-2.42.5+abi=4.0/lib/libwebkit2gtk-4.0.so.37
#4  0x00007fffe970d9a5 in IPC::Connection::dispatchMessage(std::unique_ptr<IPC::Decoder, std::default_delete<IPC::Decoder> >) ()
   from /nix/store/pjprkpwc95bxqqda4n3wrxy4w23l2fzi-webkitgtk-2.42.5+abi=4.0/lib/libwebkit2gtk-4.0.so.37
#5  0x00007fffe970f98e in IPC::Connection::dispatchIncomingMessages() () from /nix/store/pjprkpwc95bxqqda4n3wrxy4w23l2fzi-webkitgtk-2.42.5+abi=4.0/lib/libwebkit2gtk-4.0.so.37
#6  0x00007fffe56d6492 in WTF::RunLoop::performWork() () from /nix/store/pjprkpwc95bxqqda4n3wrxy4w23l2fzi-webkitgtk-2.42.5+abi=4.0/lib/libjavascriptcoregtk-4.0.so.18
#7  0x00007fffe573d6b9 in WTF::RunLoop::RunLoop()::{lambda(void*)#1}::_FUN(void*) ()
   from /nix/store/pjprkpwc95bxqqda4n3wrxy4w23l2fzi-webkitgtk-2.42.5+abi=4.0/lib/libjavascriptcoregtk-4.0.so.18
#8  0x00007fffe573e14f in WTF::RunLoop::{lambda(_GSource*, int (*)(void*), void*)#1}::_FUN(_GSource*, int (*)(void*), void*) ()
   from /nix/store/pjprkpwc95bxqqda4n3wrxy4w23l2fzi-webkitgtk-2.42.5+abi=4.0/lib/libjavascriptcoregtk-4.0.so.18
#9  0x00007fffedbf7629 in g_main_dispatch () from /nix/store/hbdl12ck2ibhn5zsi8nsfa6v471pf0j6-glib-2.78.4/lib/libglib-2.0.so.0
#10 0x00007fffedbfa797 in g_main_context_iterate_unlocked.isra () from /nix/store/hbdl12ck2ibhn5zsi8nsfa6v471pf0j6-glib-2.78.4/lib/libglib-2.0.so.0
#11 0x00007fffedbfb04f in g_main_loop_run () from /nix/store/hbdl12ck2ibhn5zsi8nsfa6v471pf0j6-glib-2.78.4/lib/libglib-2.0.so.0
#12 0x00007fffee406b85 in gtk_main () from /nix/store/rp383l10xlx4l8mqjk98zd2x8q0ilk7r-gtk+3-3.24.39/lib/libgtk-3.so.0
#13 0x00007ffff036ecf5 in wxGUIEventLoop::DoRun() () from /nix/store/5v286cibqc7xnhfwa6m5ypgn1kqrcnd1-wxwidgets-3.1.7/lib/libwx_gtk3u_core-3.1.so.7
#14 0x00007ffff08d496d in wxEventLoopBase::Run() () from /nix/store/5v286cibqc7xnhfwa6m5ypgn1kqrcnd1-wxwidgets-3.1.7/lib/libwx_baseu-3.1.so.7
#15 0x00007ffff08a974b in wxAppConsoleBase::MainLoop() () from /nix/store/5v286cibqc7xnhfwa6m5ypgn1kqrcnd1-wxwidgets-3.1.7/lib/libwx_baseu-3.1.so.7
#16 0x00007ffff0916607 in wxEntry(int&, wchar_t**) () from /nix/store/5v286cibqc7xnhfwa6m5ypgn1kqrcnd1-wxwidgets-3.1.7/lib/libwx_baseu-3.1.so.7
#17 0x000000000138160b in Slic3r::GUI::GUI_Run(Slic3r::GUI::GUI_InitParams&) ()
#18 0x00000000008f072b in Slic3r::CLI::run(int, char**) ()
#19 0x0000000000869644 in main ()
```

### Expected results

Not crashing.

### Project file & Debug log uploads

[debug_Tue_Feb_13_18_27_28_203063.log.0.zip](https://github.com/SoftFever/OrcaSlicer/files/14270161/debug_Tue_Feb_13_18_27_28_203063.log.0.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

mabl, 2024-02-13:
> I am using KDE with fractional scaling and usually apply the WEBKIT_DISABLE_COMPOSITING_MODE=1 workaround from https://github.com/SoftFever/OrcaSlicer/issues/2703. Unfortunately, this new crash appears with and without that flag.

I was wrong - the crash _does not_ happen with `WEBKIT_DISABLE_COMPOSITING_MODE=1`  - though things like the setup wizard are not properly rendered without it...

mabl, 2024-03-21:
Still crashes with `WEBKIT_DISABLE_COMPOSITING_MODE=1` on 2.0 beta.

mabl, 2024-03-21:
With `WEBKIT_DISABLE_DMABUF_RENDERER=1` from https://github.com/SoftFever/OrcaSlicer/issues/4381#issuecomment-1999439123, I can see the dialogs and don't get any crashes.

closed 2024-06-27 by github-actions[bot]

eraserhd, 2025-05-14:
This SEGV is happening to me, too.  No environment variables seem to affect it.