OrcaSlicer/OrcaSlicer #4085: Random incorrect filament coloring on second to last and third to last layer

nialtilson, 2024-02-13:
### OrcaSlicer Version

1.9.0

### OS version

macOS Ventura 13.0

### Additional system information

[image]
[image]

### Printer

Bambu Labs X1C

### How to reproduce

- Coloring the lettering (specifically L, A, N, D) white
- Coloring the unicorn as desired (see .3mf file and attached images for desired coloring)
- Slice the plate
- See the incorrect coloring on the sliced plate and model

### Actual results

[image]
[image]
[image]
[image]
[image]

### Expected results

1. Lettering - the lettering should be white with no random coloring on the second to last layer

2. Unicorn - The desired coloring on the top layer should also be mirrored/or the same along the same color boundaries on the layers below.

### Project file & Debug log uploads

[Harmony.3mf.zip](https://github.com/SoftFever/OrcaSlicer/files/14272109/Harmony.3mf.zip)

### Checklist of files to include

- [X] Log file
- [ ] Project file

dfinberg, 2024-02-20:
I don't see any slicing issues in the lettering with 1.9.1 on windows (but do in the current bambu).  The pony looks to be an issue where the volumes in the underlying object are not vertical from the visible top to the bottom, some of them (namely the object being colored in pink) are sprawling. How the object is colored it's coloring that entire box, which then bleeds through the thinner white layer above it. I don't know of any way to easily recolor that. You might be able to set the entire structure to white and then paint the top layer by hand.

nialtilson, 2024-02-20:
Hello David,

Thank you very much for your reply and for checking on this issue. I just
downloaded 1.9.1 to see what changes are now visible, and unfortunately,
nothing of note. To start, I uploaded the previous project with the full
coloring and sliced the plate - the issues on the lettering and unicorn
both appeared still. *Are there differences between the Windows and Mac
versions? If so, why? *It looks like you might be having different
experiences than me then with your version being able to remove the issue
on the lettering.

As a next step, I started fresh with the STL and re-did the coloring. First
with the stars and the background - no issue. Then with the orange of some
features on the unicorn - the lettering issue reappeared. Lastly, I removed
the coloring on the walls of the orange parts and just left the top layer -
and regardless of if I paint by using 'fill' or 'triangle', the issue on
the lettering persists. I've attached each iteration of those tests for
analysis, in case there might be any smoking gun that would clarify why any
coloring on the surface of the unicorn would influence the creation of
errant incorrect color paths on the lettering.

At this point, unless an actual root cause in the code and slicing is found
(and I hope one is!), the only option I see right now is to re-do the model
so that every different-colored feature is a separately-extruded body with
a minimal gap to the next. That would isolate each structure and isolate
each color. Hopefully that would then resolve the issue. But I'm more than
happy to keep hacking away at this to try and find the root cause in the
code with you and your teams' support!

Thank you very much for your time and support. If you have any further
questions or potential solutions for me to try, I'm happy to help. I hope
you have a great day, and I look forward to hearing from you soon!

Sincerely,

Nial

ft276, 2024-02-25:
I have a similar issue
- multi part object within step file
- bottom: petg (manual support), middle: pla (support interface), upper: petg (actual model)
- first issue is the usual: orcaslicer treats adjacent parts as contiguous meaning specifying bottom/top layers on a per part basis does nothing, mandating use of awkward top/bottom shell layer numbers and/or shell thicknesses
- second issue is that setting the number of shells appropriately causes the middle pla part to have one correct (pla, solid) layer and one incorrect (pla on walls, petg everywhere else) layer. This makes it unusable as support.

Have tried:
- setting walls to a high enough number as to avoid the need for top/bottom shell: still left a small area as petg
- setting infill to 100% which caused the same behaviour, although 99% is ok - a dirty workaround
-  changing default filament of the entire object to pla, which did nothing (prior to specifying filament type for the individual parts)
Final workaround: delete the interface support middle layer entirely, and let orcaslicer generate the support interface itself, which seems to work.
-

nialtilson, 2024-03-05:
To update on the ongoing process of this issue:

To solve the miscoloring issue, I went through and re-worked the entire model. This meant creating separate extrusions specific to each section of different color. This was a tedious and time consuming process, but after slicing the coloring issue was no longer present as noted before. This allowed me to print the model without the pre-explained incorrect coloring paths. 

The original miscoloring issue was then noted on a different model. Using the same colors and a similar mode of coloring (Fill, Sphere, and Triangle), there was another instance of this issue occuring with a stray incorrect path of coloring into a different area.

On another front, I've also submitted a ticket to BambuLab, because this issue is present in both Orca and BambuLab Slicer. At this point, the developers have recognized there is an issue but have not stated when it can/will be fixed. That issue's been listed as resolved and now it's just a waiting game.

ft276, 2024-03-05:
In my case the layer in question was already part of a separate part that was a single object throughout, so presumably a subtly different issue.

ft276, 2024-03-06:
No, definitely a bug. 
Multipart model
- a) at bottom, b) single layer above, c) single layer above, d) 3 layers on top, all with 0mm z offset
- a) is PETG, b) is PLA support, c) is PLA support (split into different objects so upper has ironing on), d) is PETG
- when a) has sparse infill at the top, all is good - note a) has sparse infill at the top even when there are 5 top layers specified, because it starts counting from the very top layer directly above it instead of the top layer of the part (another behaviour I consider a bug, but not for this thread)
By all is good, I mean PETG parts are sliced as PETG and PLA parts are sliced as PLA.
- when a) has any actual top layers, it breaks. For example, if I change from '5 top shell layers' to '6 top shell layers' - it has a single correct top layer, but it also 'invades' the layer above, reducing part b) to a couple of walls of PLA with the rest PETG. The line type stays as 'Internal Bridge' for both scenarios.

See attached photo - top two are with 6 top shells on part a), middle two are with 5 top shells on part a), bottom two are with workaround (below).
I want top shells present on part a) to provide a smoother part b) and c), to ensure a smooth bottom surface on part d), but currently I have to use a workaround of 100% walls on part b) to avoid PETG contamination.

While it's great there's a workaround I can use for the moment, it's clearly not meant to behave this way.

[image]

closed 2024-06-12 by github-actions[bot]