OrcaSlicer/OrcaSlicer #4091: Filament shrinkage is mixing an objects' coloring

Gil80, 2024-02-14:
### OrcaSlicer Version

1.9.0

### OS version

Windows 11

### Printer

Bambu Lab X1C

### How to reproduce

1. Open an object that is suitable for multi color
2. Color with two colors (black and red for example)
3. Slice the object and verify all looks well
4. Enter to one of the selected colors filament settings
5. Set the shrinkage value, in my case 99.97% for PLA
6. Slice the object again

### Actual results

The slicer mixes the coloring.

[image]

### Expected results

Filament shrinkage should not affect the object multi-coloring

[image]

### Project file & Debug log uploads

[two-faced-sturdy-infinity-cube-20.zip](https://github.com/SoftFever/OrcaSlicer/files/14279627/two-faced-sturdy-infinity-cube-20.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

Fliens, 2024-03-22:
I have the same issue

Without shrinkage:
[image]

With shrinkage:
[image]

niteowl660, 2024-03-24:
Same here

Without Shrinkage
[image]

With Shrinkage
[image]

SCISOK, 2024-03-26:
I also have this same issue.

igiannakas, 2024-06-20:
Please test this here: https://github.com/SoftFever/OrcaSlicer/pull/5774

closed 2024-08-28 by SoftFever