OrcaSlicer/OrcaSlicer #4092: Hide/View Function of Plates and Objects for them to be non-selectable whilst hidden

Johann-D, 2024-02-14:
**Is your feature request related to a problem? Please describe.**
Through rigorous Multi-Plate organizing which is Amazing in Orca, it's hard to properly adjust filled plates with roughly 10-12 single pieces/modifiers when i have to play minesweeper just to move the camera instead of objects in the far reaches of the plate galaxy.

**Which printers will be beneficial to this feature**
Everyone, because it's relevant for every plate organization progress.

**Describe the solution you'd like**
A View/Hide button in the Objects list when viewing per object settings, ideally to the left of the object in the object list. Might also be beneficial to have it for plates as well.

**Describe alternatives you've considered**
Alternatively possible to just hide plates instead, maybe under the floating "lock plate" button next to plates.

kylek29, 2024-02-14:
Yes, I agree. This has been suggested a lot on the PrusaSlicer github (alongside other UI/UX improvements in this area) as well. 

From a UI/UX workflow, I think having checkboxes for Visibility (hide/view model), Printability (enables/disables whether the slicer will process it), and Selectability (disables the user's ability to click it, but you can still use it as a reference if need-be) would be the most rounded approach.

The Selectability also doubles as a quasi-lock state, so that covers another one of the common feature requests I've seen pop up on PrusaSlicer.

closed 2024-05-23 by github-actions[bot]

Dochartaigh, 2025-11-10:
Also asking for this to be reopened. Hiding object would be SO useful it isn't even funny...