OrcaSlicer/OrcaSlicer #4093: Issue slicing model with Spiral Vase mode. Partial missing layer.

0ldHand, 2024-02-14:
### OrcaSlicer Version

2.0.0-dev

### OS version

Windows 11

### Printer

Bambu P1S

### How to reproduce

Slicing full model in Cura 4.6.1 is successful in spiral mode with no missing layers or seams. (Eclipson Helix plane model)
Slicing same model in Orca produces both a seam and missing lines.
Including relevant portion of stl.
[Fuselage_F1-Spriral-Testing.zip](https://github.com/SoftFever/OrcaSlicer/files/14282210/Fuselage_F1-Spriral-Testing.zip)

### Actual results

Orca does not appear to be able to generate an uninterrupted tool path in spiral mode with this model. 
[image]

### Expected results

Orca produces a missing section 1 layer thick on part of the model and creates a seam. I can probably get permission to share the entire stl from Eclipson if needed.
[image]

### Project file & Debug log uploads

[Fuselage_F1-Spriral-Testing.zip](https://github.com/SoftFever/OrcaSlicer/files/14283251/Fuselage_F1-Spriral-Testing.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

pollosud, 2024-02-15:
I have the same problem. When I pront in Vase Mode, after about 10-12 layers, it does not print 2 layers .

Noisyfox, 2024-02-18:
Does this issue occur on 1.9?

closed 2024-05-27 by github-actions[bot]

mduclehcm, 2024-10-07:
I got the same issue on version 2.1.1 when printing rc air place in vase mode. The model is sliced perfect in cura slicer.

DBShrill, 2024-11-27:
Same issue. Printing a spiral vase wing from **https://www.youtube.com/watch?v=QJjhMan6T_E** and have random empty layers. Only remedy is to increase "Slice Gap Closing Radius" until the empty layers disappear but everything else prints normally.
<img width="1173" alt="Screenshot 2024-11-27 at 5 09 39 PM" src="https://github.com/user-attachments/assets/13773e52-1837-4f64-a7a2-
[image]
3295c05d05d1">