OrcaSlicer/OrcaSlicer #4099: Build fails on raspberry pi, "Error 2"

Ender436, 2024-02-15:
### OrcaSlicer Version

1.9.0 (I think, I just cloned the repository)

### OS version

Latest version of Raspbian

### Additional system information

Raspberry pi 5 8GB ram version

### Printer

none

### How to reproduce

Follow the Linux build instructions on a raspberry pi

### Actual results

Everything seem to build fine for a while, until at 65% the build outputs those lines after finishing building dep_OCCT. 

make[1]: *** [CMakeFiles/Makefile2:738: CMakeFiles/deps.dir/rule] Error 2
make: *** [Makefile:423: deps] Error 2

### Expected results

It should continue to build

### Project file & Debug log uploads

Don't see any generated logs

### Checklist of files to include

- [ ] Log file
- [ ] Project file

DrKlipper, 2024-02-19:
I think this is a RAM problem. 
[image]
Tried it on my Pi 4 with 8GB and the RAM maxed out.

DrKlipper, 2024-02-21:
Hi !
So my results are as following: 

- Compiling on a Pi needs at some points ~12-13GB Ram. So I ended up adding 8GB of SWAP to the system, which works fine for me. 
- Compiling on a Pi 4 takes about 4-5 hours! It takes forever :-) 
- Running Orca on a Pi 4 is fine until you slice something and want to make changes to the rendered view (like rotating, panning, ...) This is nearly impossible. I suppose it's due to missing OpenGL Support ? So using Orca on a Pi 4 is a no-go at this point. Don't know if there are any compile options to use some other 3D acceleration ? 
- Next try for me will be a Pi 5. The soc is ~2-3 times faster. So compiling should be quicker and hopefully it's useable after compiling. 

Hope this helps :-)

Ender436, 2024-02-21:
Thanks for all the help! I thought I'd just try running the build again after seeing your comment, and for some reason it decided to work that time, it did crash a couple times during the build, but I just ran the command again and it seemed to pick up where it had left off. Now the only problem I'm facing is that the projects page and the setup wizard are both unusable. It seems like they're both rendering, but the pixels are being offset or something, because the colors are present, just garbled.

Ender436, 2024-02-22:
Running "export WEBKIT_DISABLE_COMPOSITING_MODE=1” and then running the appimage seemed to have fixed it.

closed 2024-05-29 by Ender436