OrcaSlicer/OrcaSlicer #4107: Linux version issues

anh977, 2024-02-16:
### OrcaSlicer Version

1.9.0

### OS version

Ubuntu 20.04

### Additional system information

Gforce GTX 460

### Printer

Klipper Generic

### How to reproduce

For me, just start it up.  I've tried 1.7.0, 1.8.0 and 1.8.1.  These simply wind up with a blank "untitled screen" and lock up. 1.9.0 starts fine but shows no build plate.  I've fiddled with settings, changed to Creality printers, no difference.  The only issues I've seen in the logs are "Unable to create exclude triangles" and "Unable to open the .stl" file I tried to import. There are no permission issues and if I tell it to save the project, it writes it alongside the .stl file.

### Actual results

No buildplate shown but the program acts like its there.

### Expected results

I should see the stl rendered on a visible buildplate.

### Project file & Debug log uploads

[Report.zip](https://github.com/SoftFever/OrcaSlicer/files/14305320/Report.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

anh977, 2024-02-17:
Same thing with 1.9.1

kabutobelial, 2024-02-25:
The same thing happens to me, I can't see the buildplate. I have Ubuntu 22.04 GTK3

dasmikko, 2024-04-02:
I'm not sure if you guys have the same issue, but i'm using issue just in case.

When I run Orca Slicer on Fedora 39 with a Nvidia GPU, I get this error message in the terminal: 
`AcceleratedSurfaceDMABuf was unable to construct a complete framebuffer`

It results in the wizard being blank, and various parts of the UI.

I can bypass this issue with the following:

`WEBKIT_DISABLE_DMABUF_RENDERER=1` and to make it permanent, add it to `/etc/environment`

I'm not sure what the consequences of disabling the DMABUF renderer, but it make the application work for me.

nick-laird, 2024-04-02:
> I'm not sure if you guys have the same issue, but i'm using issue just in case.
>
> When I run Orca Slicer on Fedora 39 with a Nvidia GPU, I get this error message in the terminal: [...]

You shouldn't set it in your environment file like that unless you want it to disable for your entire system. you can set it at runtime by adding it to the Appimage shortcut or putting it before the executable on the shell. 

This thread talks about the issue, and a fix is to set the environment variable like they mention. It's the webkit2gtk library causing the trouble with NVIDIA drivers. So it's technically not Orca Slicer's bug but a library it uses. 
https://bugs.webkit.org/show_bug.cgi?id=261874

The workaround that has been rock solid for me is setting WEBKIT_DISABLE_DMABUF_RENDERER=1 as an environment variable before launch. It doesn't impact my whole system then and Webkit can keep using DMA in places it still works.

dasmikko, 2024-04-03:
> You shouldn't set it in your environment file like that unless you want it to disable for your entire system. you can set it at runtime by adding it to the Appimage shortcut or putting it before the executable on the shell.

Yeah, I figured that. I just made a small script that executes the appimage with the variable set.

closed 2024-07-10 by github-actions[bot]