OrcaSlicer/OrcaSlicer #4114: Can't save new filament

Banner-Keith, 2024-02-16:
### OrcaSlicer Version

1.9.0

### OS version

Windows 11 Pro 22H2

### Printer

Sovol SV06 Plus

### How to reproduce

1. I open a brand new project in Orca
1. Using my `Sovol SV06 Plus 0.4 nozzle - Custom` printer profile.
1. I have the `Sovol Generic PETG - Z Offset` filament selected. 
1. Edit Filament
1. Don't change anything. Just hit save and type in the new name "Overture PETG" as User Preset.
1. Save changes to filament settings.
1. Application stops responding and eventually has to be killed via Task Manager.

### Actual results

[image]

### Expected results

New filament should save.

### Project file & Debug log uploads

[NewFilamentCrashProjectBlank.zip](https://github.com/SoftFever/OrcaSlicer/files/14314604/NewFilamentCrashProjectBl
[debug_Fri_Feb_16_11_16_38_35756.log.0.zip](https://github.com/SoftFever/OrcaSlicer/files/14314658/debug_Fri_Feb_16_11_16_38_35756.log.0.zip)
ank.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

mordelon, 2024-02-23:
Hello
It happens to me exactly, I can save only if I don't change the name of the profile. (Save it as a copy)

I am in Windows 10 and Orcaslicer 1.9.1
CR6 printer and mingda magician x

Buster-Grob, 2024-02-25:
Same thing happens to me. 

Windows 11, OS 1.9.1
RatRig VCore3.1 500

[log.zip](https://github.com/SoftFever/OrcaSlicer/files/14398100/log.zip)

mordelon, 2024-02-26:
I am testing version 2.0 DEV and apparently this problem is not presented.

SuhbatDaRobot, 2024-04-15:
I have the same issue, started presenting today.

closed 2024-07-22 by github-actions[bot]