OrcaSlicer/OrcaSlicer #4118: Filament Change (M600) should be added after nozzle is already in position

manferlo81, 2024-02-17:
### OrcaSlicer Version

1.9.0 & 2.0.0 dev

### OS version

windows 10

### Additional system information

Core i7, 8Gb ram... it doesn't really matter for this issue

### Printer

Ender 3

### How to reproduce

Add a color change

### Actual results

Orca adds the M600 command at the end of previous layer, which causes the nozzle to return to that position after the filament is added. When the nozzle is returning from the "parking"position it usually oozes some filament then it goes to the print and drag that filament to the next layer start position, sometimes creating an undesired string or blob.

### Expected results

The nozzle should move to the start position of the next layer first, then add the M600, the nozzle parks, the filament is changed and the nozzle returns to the position it was before, in this case it's already at the start position of the layer, and if it oozes a little bit it doesn't really matter as in most cases it will be inside the part, assuming you are printing inner wall or infill first, which most people are.

I believe other slicer recently implemented something like this I read, I believe it was Prusa's

### Project file & Debug log uploads

[M600 String Issue.zip](https://github.com/SoftFever/OrcaSlicer/files/14318071/M600.String.Issue.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

manferlo81, 2024-02-17:
As an alternative an option can be added to let the user decide between the previous behavior and the new one.

luco85, 2024-02-26:
i have same issue

tkunchick, 2024-02-29:
Might be helpful to list what firmware you are using. Marlin? Klipper? Something else?

If Klipper, this seems like a macro issue rather than slicer.

luco85, 2024-02-29:
I use clipper. look in the code windows. 
[image]
the final zhop is after M600. After this it return to last point to execute zhop on the last point. If you use prime tower all works fine.
[image]

tkunchick, 2024-02-29:
If you are using manual filament changes and not a tool changer or MMU, maybe try to replace the m600 with a standard PAUSE and see if that cause the same issue.

tkunchick, 2024-02-29:
Also looks like you are printing external walls first.

```
12839 G1 E2 F1200
**12840 ;TYPE:Outer wall**
12841 G1 F1500
```

luco85, 2024-02-29:
> Also looks like you are printing external walls first.
> 
> ```
> 12839 G1 E2 F1200
> **12840 ;TYPE:Outer wall**
> 12841 G1 F1500
> ```
never used

[image]

luco85, 2024-02-29:
> If you are using manual filament changes and not a tool changer or MMU, maybe try to replace the m600 with a standard PAUSE and see if that cause the same issue.

yep same issue with pause. zhop is after pause command and next filament load (set pressure advance...)

vgdh, 2024-03-12:
I've found a workaround. Enable **purge tower** with purge amount of 1mm3 and use **multiplier** of 0 in **Flushing volumes**. 
This effectively creates minimal tower for just the positioning the head.

luco85, 2024-03-12:
> I've found a workaround. 
> 
> 1. Enable **purge tower** 
> 
> 2. In filament settings choose purge amount of 1mm3 
> 
> 3. Use **multiplier** of 0 in **Flushing volumes**. 
> 
> 
> 
> This effectively creates a minimal tower for just the positioning the head.

Yeo i already use similar metod, but i hope they fix the issue

awabom, 2024-06-05:
FYI: This has been fixed in PrusaSlicer, see: https://github.com/prusa3d/PrusaSlicer/issues/2672
(Also, there is a workaround Python script in that ticket until this is fixed)

closed 2024-09-12 by github-actions[bot]