OrcaSlicer/OrcaSlicer #4132: Linux Mint latest version

Vance1122, 2024-02-17:
### OrcaSlicer Version

1.9.1

### OS version

Mint based on Ubuntu 22

### Printer

X-Max 3

### How to reproduce

Start the app.

### Actual results

I get a set up wizard page that is blank. Can not proceed to set up any printer.
[image]

### Expected results

I should see a populated window with options to select so I can set up the slicer.

### Project file & Debug log uploads

There are no log files as I haven't even gotten to use the program.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

darco415, 2024-02-18:
Strange, works fine here on LM...I was on other versions so assuming it just copied over profile. Did you have OS installed prior to this upgrade?

Vance1122, 2024-02-18:
Yes. Used to work fine in version before 1.8. But when I tried 1.9.1, I get
nothing.

Vance1122, 2024-02-18:
This is a fresh install of Linux Mint and I can't even get to the point of bringing in profile back ups. Also when you see what I see on the screen, orca slicer can not be shut down or exited. It is stuck like that until you reboot.

karolbe, 2024-02-21:
I have the same problem as @emelbardis also Fedora 39.

darco415, 2024-02-24:
I remember what I did months ago, when OS was having this similar issue, I reverted to Nouveau drivers instead of Nvidia

Vance1122, 2024-02-25:
I reverted to the open source video driver as mentioned and I was finally able to see the start up wizard. I set up all the printers I want to use this slicer with.  I closed the program. Then later I went to start it and it takes 3 minutes for the program to start. This is frustrating.

Vance1122, 2024-02-25:
[Orcaslicer start.txt](https://github.com/SoftFever/OrcaSlicer/files/14395327/Orcaslicer.start.txt)
Here is a copy and paste of the output when starting the slicer via the terminal. Hopefully that will help?

darco415, 2024-02-27:
whats the command? Ill run it here for comparison

Vance1122, 2024-02-27:
In a terminal window, navigate to the directory you have put the appimage. Then enter this:  ./OrcaSlicer_Linux_V1.9.1.AppImage

That will cause the program to start.

Vance1122, 2024-02-27:
> OS output.txt

Seems to be the same errors.

darco415, 2024-02-27:
except for this:
(orca-slicer:6271): Gtk-CRITICAL **: 21:24:50.698: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed
KMS: DRM_IOCTL_MODE_CREATE_DUMB failed: Permission denied
Failed to create GBM buffer of size 1200x704: Permission denied
KMS: DRM_IOCTL_MODE_CREATE_DUMB failed: Permission denied
Failed to create GBM buffer of size 1200x704: Permission denied
KMS: DRM_IOCTL_MODE_CREATE_DUMB failed: Permission denied
Failed to create GBM buffer of size 1200x704: Permission denied
Failed to create EGL images for DMABufs with file descriptors -1, -1 and -1
KMS: DRM_IOCTL_MODE_CREATE_DUMB failed: Permission denied
Failed to create GBM buffer of size 1200x740: Permission denied
KMS: DRM_IOCTL_MODE_CREATE_DUMB failed: Permission denied
Failed to create GBM buffer of size 1200x740: Permission denied
KMS: DRM_IOCTL_MODE_CREATE_DUMB failed: Permission denied
Failed to create GBM buffer of size 1200x740: Permission denied
Failed to create EGL images for DMABufs with file descriptors -1, -1 and -1
KMS: DRM_IOCTL_MODE_CREATE_DUMB failed: Permission denied
Failed to create GBM buffer of size 1200x740: Permission denied
KMS: DRM_IOCTL_MODE_CREATE_DUMB failed: Permission denied
Failed to create GBM buffer of size 1200x740: Permission denied
KMS: DRM_IOCTL_MODE_CREATE_DUMB failed: Permission denied
Failed to create GBM buffer of size 1200x740: Permission denied
Failed to create EGL images for DMABufs with file descriptors -1, -1 and -1

Vance1122, 2024-02-28:
Sounds like a digital rights management problem. But all I did was download the appimage and gave it permission to run like a program. What is all that KMS_DRM stuff?

darco415, 2024-02-28:
maybe webkit related?

sylvainbarriere, 2024-03-24:
Same issue on kubuntu 22.04

Vance1122, 2024-03-24:
If you are running proprietary drivers for your display/graphics card, then
this happens. Change to the open source drivers and orca slicer will work
properly.

karolbe, 2024-03-24:
But...the thing is that no one will switch to (rubbish) Nvidia open source drivers just to run Orca slicer...There has to be other explanation.

Vance1122, 2024-03-24:
I know. I switched to the open source drivers, set up all my printers in
orca, then switched back to the proprietary drivers and it seems to be
working.  Just a pain in the rear end if you want to add a new printer.

sylvainbarriere, 2024-03-25:
Thanks Vance1122, it's a bit painful, but I will finally be able to try orcaslicer!

Edgardo1969, 2024-05-13:
Hello, I am not a programming expert but I am a very patient person.
I found the conflict that the Orcaslicer app has on Linux mint, it is the JavaScript update "webkit2gtk - 2.44.0-0ubuntu0.22.04.1".
Avoid installing that version of Java and wait for a fix.
Note: if anyone knows how to uninstall that version, please pass on the method.
[image]

Edgardo1969, 2024-05-13:
Octoprint should not be installed, it comes with a Java update and corrupts Orcaslicer functions.

When installing these updates or programs that contain Java and update it, most likely to use Orcaslicer, you will have to reinstall Linux Mint from 0. That happened to me.

Vance1122, 2024-05-13:
For me it was the graphics card driver. If I use the generic open source
driver, Orca Slicer works fine. If I use NVidia's proprietary driver, then
I can not access settings  or set up a new printer in Orca Slicer.  So I
change to the generic driver and make my changes in orca slicer, then
switch back to the proprietary driver. I can use orca slicer with the
proprietary driver, just can't make any settings changes.

tcables, 2024-09-27:
Found that you can run:

`WEBKIT_DISABLE_COMPOSITING_MODE=1 /PathToAppImage`

in a shell.

Works for me with latest stable build 2.1.1

Thanks to Post by [ahrifox950](https://forums.linuxmint.com/memberlist.php?mode=viewprofile&u=383068)

https://forums.linuxmint.com/viewtopic.php?p=2516253&sid=62f3db3956c845404ce782301220421d#p2516253

SmileyChris, 2024-10-09:
Thanks, `WEBKIT_DISABLE_COMPOSITING_MODE=1` worked for me too.

kaips1, 2026-03-17:
why is still an issue, still cant use nvidia drivers with orca, why is always one thing after another, this used to run so smoothly