OrcaSlicer/OrcaSlicer #4133: Staggered layer perimeters

shortbus, 2024-02-18:
https://www.youtube.com/watch?v=5hGm6cubFVs

CNC Kitchen
Brick Layers: Could printing STRONG parts be so simple?

Hammerfest, 2024-02-18:
Thats pretty insane, 10% while seemingly like nothing is huge at scale.

Hopefully someone reaches out to work through some of the issues he was unable to tackle to make it a usable method.

Noisyfox, 2024-02-18:
That's interesting. I'll try implementing this when I got some time.

paulnaude, 2024-03-17:
Very smart idea indeed. It falls in my "Why didn't I think of that?" category!
His visuals also show (without him saying it) why a little bit of increased flow will be a great idea to fill those gaps between the layer lines for maximum layer surface adhesion.

AbitNutzT00, 2024-04-15:
I would really appreciate it if this could be implemented. Is this actively being worked on?

balloch, 2024-05-31:
@Noisyfox I'm happy to contribute as well, where are you so far on this?

cochcoder, 2024-05-31:
There is a issue request on [PrusaSlicer](https://github.com/prusa3d/PrusaSlicer/issues/1823) and they have come up with some cool ideas like [this one](https://github.com/prusa3d/PrusaSlicer/issues/1823#issuecomment-1961309760).

closed 2024-09-06 by github-actions[bot]

doudar, 2024-10-21:
This is a really great idea. Hopefully someone can find time to implement it.

theboee, 2024-10-27:
@Noisyfox Bumping this. Really want to see this added to a slicer.

AbitNutzT00, 2024-10-27:
Me, too. It doesn't sound like they're interested, overwhelmed, have other priorities, or all of the above.

xboxhacker, 2024-11-08:
BUMP. I need this!!

jside86, 2024-11-08:
New video from Geek Detour on this:

https://www.youtube.com/watch?v=9IdNA_hWiyE

Apparently, someone was able to get a patent in 2020 that suggest the same thing as a previous Stratesys Patent from 1995. The patent should not have been approve and would most likely not hold up in court. 

Someone should create a update for Orca for this feature alone. 

Imagine having stronger layer adhesion!

Nebur99, 2024-11-09:
> New video from Geek Detour on this:
The guy also summarized all the relevant facts in text: https://www.patreon.com/posts/115566868

Shall0e, 2024-11-14:
Bumping this as well! We need this.

And yes, I did come from Geek Detour.

sudokit, 2024-12-15:
This would be such a cool feature if implemented

roger181078, 2025-01-23:
Testing is now possible:

[https://www.youtube.com/watch?v=EqRdQOoK5hc&list=LL&index=2](url)

[https://github.com/TengerTechnologies/Bricklayers](url)

Again, many thanks to the person who made this possible, and shared it with the community.