OrcaSlicer/OrcaSlicer #4136: Paint Text

KevinFairgrieve, 2024-02-18:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Which printers will be beneficial to this feature**
E.g. Voron/Klipper based printer/etc...

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

I am using a Bambu Lab P1S and would like to be able to paint text. When painting on a flat surface I would like to be able to select paint and then as an option in tool type. "TEXT".

Noisyfox, 2024-02-19:
Use text tool as a modifier and set the filament.

KevinFairgrieve, 2024-02-19:
> Use text tool as a modifier and set the filament.

Apologies, really new to Orca slicer. Could you explain in simple steps.

Many thanks.

Noisyfox, 2024-02-20:
Right click the main object that you want to paint text on, select "Add modifier"->"Text", then put what ever text you want. Then go to object list, change the filemant you want to use for the text
[image]

KevinFairgrieve, 2024-02-20:
Got it, Thank you. 

Closing the issue.

closed 2024-02-20 by KevinFairgrieve