OrcaSlicer/OrcaSlicer #4139: TempTower non-manifold in 1.9.x

WimVDK, 2024-02-18:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

1.9.1

### Operating System (OS)

Windows

### OS Version

11 23H2

### Printer

Bambulab X1C

### How to reproduce

1. Click Calibration
2. Click Temperature
3. Pick any tower type
4. Tower gets added

### Actual results

Loaded temp tower is non-manifold
[image]

### Expected results

Manifold tower should be loaded. Towers in 1.8.x were fine.

### Project file & Debug log uploads

[Nozzle temperature test.zip](https://github.com/SoftFever/OrcaSlicer/files/14323043/Nozzle.temperature.test.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

Sp4rkR4t, 2024-03-12:
Just found this issue when testing 2.0
[image]

WimVDK, 2024-03-31:
Issue persists in 2.0.0

Towers still not manifold

closed 2024-07-08 by github-actions[bot]

hho, 2025-01-16:
Still an issue in 2.2.0