OrcaSlicer/OrcaSlicer #4145: Changing printer selection clears out Process overrides

Chewieez, 2024-02-19:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

1.9.1

### Operating System (OS)

macOS

### OS Version

14.3.1

### Printer

Bambu Labs P1S

### How to reproduce

1. Download a profile from MakerWorld which is setup on another printer other than P1S, such as X1 and also has some Process overrides such as speed, quality or strength changes. (even if profile shows support for P1S)
2. Open 3mf file in OrcaSlicer
3. Change the printer selected to the P1S
4. Notice that many of the Process overrides are removed

### Actual results

If the profile you've downloaded is set for say a X1 and has some Process overrides, they are cleared and are not retained after changing the selected printer. 

I performed these same steps using Bambu Studio 1.8.4.51 and this bug does not happen, and all of the Process overrides are retained when changing the selected printer. 

### Expected results

The Process overrides should be retained after changing the selected printer

### Project file & Debug log uploads

[The+Ultimate+SPOOL-X1.3mf.zip](https://github.com/SoftFever/OrcaSlicer/files/14325780/The%2BUltimate%2BSPOOL-X1.3mf.zip)
[debug_network_Sun_Feb_18_11_58_42.log.enc.zip](https://github.com/SoftFever/OrcaSlicer/files/14325782/debug_network_Sun_Feb_18_11_58_42.log.enc.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

gshpychka, 2024-02-20:
I can confirm that this happens to me all the time, and I'm also on MacOS.

MotoInsight, 2024-02-21:
I can also confirm this is happening repeatedly to me as well, on MacOS. This behavior does not happen in BambuSlicer.

I literally registered just to add this comment as it's so annoying.

MotoInsight, 2024-02-24:
Additional follow-up as I've been experimenting. 

- This occurs on both Mac 1.9.1 and Windows 1.9.1
- It occurs as far back as version 1.8.0 on both platforms as well.
- It also occurs (as expected) in the most recent nightly build.

TinyMito, 2024-02-27:
Yes, I am here to confirm with videos. OrcaSlicer 1.9.1 is currently unusable if I want to retain the same preset from the designer just by switching printer. It will clear all settings.

Bambu Studio Expected Behaviour.
https://github.com/SoftFever/OrcaSlicer/assets/75095713/780e5558-1b7d-446a-b66a-c3979274255d

OrcaSlicer 1.9.1 Unexpected Behaviour.
https://github.com/SoftFever/OrcaSlicer/assets/75095713/ae0876ff-b9ec-4ad2-ac66-c19071ae359c

TinyMito, 2024-03-12:
Beta 2.0.0 seems to be fixed, can someone confirm this?

MotoInsight, 2024-03-14:
This does not appear to be fixed in 2.0.0 Beta, at least MacOS ARM 64

TinyMito, 2024-03-14:
> This does not appear to be fixed in 2.0.0 Beta, at least MacOS ARM 64

That is correct, I found more details on what is happening.

1. Switch between different bed printer will trigger dialog transfer parameters (no issue)
2. Switch between same bed printer is missing, the dialog and the parameters wipes clean (bug).

Demo Video (near the end, where the issue is):

https://github.com/SoftFever/OrcaSlicer/assets/75095713/6fc7676f-f303-46bc-96ce-4443a939a9c5

@SoftFever sorry for the ping, we haven't seen any update on this reproducible bug. It appears to be between X1 and P1 share the same bed: if switched, it will wipe all parameters.

closed 2024-03-15 by SoftFever

SoftFever, 2024-03-15:
Thank you all.
The issue is fixed.
It will be included in the nightly release as well as in the 2.0.0 official release

tbitz007, 2024-05-14:
I am using OrcaSlicer 2.0.0 on a MacOS Monterey 12.7.4 and still see this issue.

I have the printer set to "Bambu Lab A1 mini 0.4 nozzle" under system presets.
I have created a user process with many changes to defaults (line width, speed, etc..)

When I switch the printer to "Bambu Lab A1 mini 0.6 nozzle" which is the same printer just larger nozzle, the process automatically switches to a system process and I can no longer see my user process.

Note the filament user process is still available between the two printers.

mgjv, 2024-06-27:
This issue still seems to exist at least in some cases. When switching to a user preset machine, all the settings for materials get reset to a single material (K1C with my network configuration -- and yes the machine preset is configured as a multi-material one). However, when changing to a built-in preset (the standard K1C configuration that comes with Orca), only the process presets get reset. The material doesn't _fully_ get reset. Just the individual materials.

OrcaSlicer 2.1.1.

Downloaded this 3mf file: https://www.printables.com/model/902136-angle-finder-gauge/files.

To reproduce, open above file, and change it to stock K1C definition. Observe the changes made. Now save a user preset of the K1C, and reopen the file, nd change to that. Observe the material information is now all lost.

rdbischoff, 2024-06-29:
Still a problem in 2.1.1. pointless to share 3MF files if when I switch to my printer it erases ALL of the benefits of a 3MF file. it is just an STL at that point.

hugemushroom69, 2024-07-01:
I can also confirm that this is still an issue in 2.1.1.

kemper, 2024-09-06:
Can also confirm it's still happening in 2.1.1. I also commented on the other (newer) issue.

malventano, 2024-09-27:
Unsure if this is a bug or intended but this workaround fixes it:
Preferences > Presets > 'Remember printer configuration' (uncheck)

evilDave, 2025-02-28:
Still a problem in 2.3.00-beta build dc49116
(in addition, value of 'Remember printer configuration' seems to have no effect - global process settings discarded, however object process settings are kept)

stnchk, 2026-04-07:
today this behaviour is still here and very frustrating me why I need to google and make extra settings to make obvious things

PR #4506 (2024-03-15): Fixed an issue that changing printer selection clears out Process overrides
Fixes #4145