OrcaSlicer/OrcaSlicer #4146: Timelapse option missing

Brakisshuro, 2024-02-19:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.0 dev

### Operating System (OS)

Windows

### OS Version

Newest 11

### Printer

K1 Max

### How to reproduce

Others - Special - Timelapse is missing. 

### Actual results

If I search it, it shows up but it is not there when I click on it. It will take me to the others tab but no option. See attached pictures

[image]
[image]

### Expected results

Timelapse option should be available

### Project file & Debug log uploads

[debug_Sun_Feb_18_11_16_39_14884.log](https://github.com/SoftFever/OrcaSlicer/files/14324520/debug_Sun_Feb_18_11_16_39_14884.log)

no need for project as it is not dependent on model.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

mohalsa, 2024-02-21:
I second this. I have the latest version on Mac Os and I can't find the timelapse option anywhere. I am using klipperized 3d printer with a USB camera installed and working perfectly in the browser Klipper interface.

jvarrone, 2024-04-06:
Ditto here.

Dweinbach, 2024-04-09:
Same here. it seems the standard timelapse is always enabled. If true, maybe show the control without only one option and add a tooltip explaining.

jvarrone, 2024-04-09:
Where is the TL video saved, and how is it named?
Thanks

Snelso91, 2024-04-23:
+1 to this request.

I want to be able to enable the smooth timelapse option in Orca Slicer for a BIQU B1 printer which has been converted to Klipper. Although TIMELAPSE_TAKE_FRAME in the timelapse g-code section in the printer options will take care of parking the nozzle and taking the frame, it does not take care of the nozzle leaking during the nozzle park, which is why I need some kind of prime tower to wipe the ooze on and then prime the nozzle on after the frame has been taken, just like how it does on the Bambu printers when using the smooth timelapse option.

I spent ages trying to find anything in the Bambu printer profile json, model definition json, etc. files that enables this special timelapse menu before I realised that it looks like it's hard-coded into Orca to only show the timelapse menu if the printer is a Bambu Lab model?
https://github.com/SoftFever/OrcaSlicer/blob/0369a14ab8a9026d3fc3f4bcdd9bd22424ca0ca5/src/slic3r/GUI/ConfigManipulation.cpp#L749

As far as I know, the timelapse gcode from the printer settings is always inserted regardless of if timelapse is traditional or smooth. Therefore, on top of inserting the timelapse gcode traditional timelapse mode just means don't insert a special timelapse prime tower, and smooth mode means do insert the prime tower after the timelapse gcode?

Given this, I'm not sure why we can't have timelapse options for non-Bambu Lab printers if it's just inserting or not inserting a prime tower after the separately defined timelapse gcode. And even if it is doing some extra stuff specific to Bambu printers, can we at least have it work for other printers as I outlined above?

awabom, 2024-06-11:
+1. I'd like this option to be visible for all printers so I can do extra parking code when it is enabled for my SV08. The 'timelapse g-code' is always added, currently.

mrkylegp, 2024-08-07:
+1 I'm on 2.1.1 for Windows and while it appears in search, the timelapse option doesn't exist. I went to check it out because the "tips" pop up told me about it. A bit silly if the feature is not there.

udenfox, 2024-10-08:
Still having this issue on 2.2.0 beta. No option for me. MacOS

re1neke, 2025-01-09:
Same issue still exists, 2.20 MacOS

closed 2025-04-17 by github-actions[bot]

Snelso91, 2025-04-21:
This "bug" still exists in that the smooth timelapse option is still restricted to Bambu Lab printers from what I can see. Maybe it would be better to mark this issue as a feature request rather than a bug, but either way it has still not been fixed/implemented in the current Orca version (2.3.0), so I wouldn't say this is resolved yet.

HermesHonshappo, 2025-06-10:
Same here.
I'd love to have a toggle available, so that I can easily enable or disable it per print.
Right now I have to adjust the layer change g-code each time I want to enable it (and most of the time I forget about it)

sizzname, 2025-07-10:
Same issue.
Timelapse aviable only for Bambu Lab printers(

[image]

ssouthurst, 2025-08-30:
+1 Same for me.  Using Octoprint - would love to be able to use the "machine" GCode to trigger the timelapse rather than the hit and miss "Layer change".