OrcaSlicer/OrcaSlicer #4163: Purge filament change into skirts.

missile150, 2024-02-20:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

For those of us who have single extruder machines, the filament change purge can be an issue due to blobs forming from purging. In Prusa Slicer, you can turn off the purge tower and run extra skirts so your filament will purge in into the skirt. For example, if you have 3 colors to change, and they will take 3 layers to make all the changes, you can set your skirt to 6 and the skirt height to 3, and Prusa will purge each color for 2 of the 6 skirts to flush the lines, and then it will do that for the 3 layer heights until the filament changes are done.

You have to be aware of the changes and layer heights these changes are happening on to get the settings correct. But it is a good way to do filament changes without blobbing. I would like to see Orca Slicer take that feature from Prusa and incorporate that feature.

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

With a M600 filament change, the new filament can use the skirt to purge instead of  purge or wipe tore.

### Describe alternatives you've considered

There are no alternative workarounds in Orca.

### Additional context

I can do this in Prusa, but am unable to produce the same results in Cura.

gshpychka, 2024-02-20:
With Core XY, would it mean raising the bed all the way to the first layer and purge there? The model can hit the top of the enclosure if it's present.

missile150, 2024-02-20:
I don't run core xy printers, but I don't see how a purge line would have any effect on the model/bed height if it purges on the current layer print.  It does not adjust the z-height while doing this.

missile150, 2024-02-20:
> With Core XY, would it mean raising the bed all the way to the first layer and purge there? The model can hit the top of the enclosure if it's present.

I don't run core xy printers, but I don't see how a purge line would have any effect on the model/bed height if it purges on the current layer print. It does not adjust the z-height while doing this.

gshpychka, 2024-02-20:
> I don't run core xy printers, but I don't see how a purge line would have any effect on the model/bed height if it purges on the current layer print. It does not adjust the z-height while doing this.

hm, but a skirt is only printed on the first layer. I guess I don't understand what you're referring to when you say "skirt".

missile150, 2024-02-20:
> hm, but a skirt is only printed on the first layer. I guess I don't understand what you're referring to when you say "skirt".

A skirt is run as high as you want.  You set the skirt loops, distance from print and layer height in the slicer settings.  You can make it as tall as the print without issue.

gshpychka, 2024-02-22:
> > > > With Core XY, would it mean raising the bed all the way to the first layer and purge there? The model can hit the top of the enclosure if it's present.
> > > 
> > > 
> > > I don't run core xy printers, but I don't see how a purge line would have any effect on the model/bed height if it purges on the current layer print. It does not adjust the z-height while doing this.
> > 
> > 
> > hm, but a skirt is only printed on the first layer. I guess I don't understand what you're referring to when you say "skirt".
> 
> It is exactly what you are thinking. However, a skirt can be run as high as you want. You set the skirt loops, distance from print and layer height in the slicer settings. You can make it as tall as the print without issue.

Hm, what would be the point of such a skirt? Why not just a prime tower?

mike-tesch, 2024-04-26:
>Hm, what would be the point of such a skirt? Why not just a prime tower?

They are generally used to "enclose" your print. It can be effective at keeping the temperature high inside the skirt when printing with material like ASA or ABS. It also helps to block any environmental airflow from hitting the part and causing warping.

xBoKoBx, 2024-06-06:
Hi
I am looking for the exact same thing.
A purge line or skirt in relation to manual filament changes (effectively a pause / resume) is greatly needed.
In addition it should be possible to make it not return to the last place it printed before moving to the new coordinates.
The slicer could generate a sacrificial place outside of print area that it moves to before pause is initialized, so it will return to this as the stored coordinate when doing resume.

closed 2024-09-12 by github-actions[bot]