OrcaSlicer/OrcaSlicer #4167: Add part, Add negative part, Add modifier | Display problem

olcayoren, 2024-02-20:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

1.9.1

### Operating System (OS)

macOS

### OS Version

14.2.1 Sonoma

### Additional system information

Macbook M1 Arm

### Printer

Elegoo Neptune 4 Pro

### How to reproduce

1- Add cube
2- Cube position center
3- Add modifier --> Cube
4- Modifier cube position center
5- Modifier cube position z-->15

### Actual results

Display problem

https://github.com/SoftFever/OrcaSlicer/assets/33813227/2a1f4d13-a563-417e-9b8c-d52f0197997c

### Expected results

None

### Project file & Debug log uploads

None

### Checklist of files to include

- [ ] Log file
- [ ] Project file

foreachthing, 2024-02-20:
That's "normal" [z- or depth fighting](https://en.wikipedia.org/wiki/Z-fighting). Unless I don't understand your issue?!

olcayoren, 2024-02-20:
Color fluctuations seemed like a problem to me, I checked it on Prusa and it was the same situation.

closed 2024-02-20 by olcayoren