OrcaSlicer/OrcaSlicer #4176: When using the Bambu Login, if you attempt to use the SSO authentications such as google the programs crashes before it is able to receive the token response from the browser

sdjrdriver, 2024-02-21:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

1.9.1

### Operating System (OS)

Windows

### OS Version

Windows 10

### Printer

Bambu Lab A1

### How to reproduce

1. Open OrcaSlicer
2. Click Login in upper left
3. Select Google as the login type without typing in a user/pass and regardless of accepting terms.
4. Takes you out to google authentication but orcaslicer closes with no error codes that I can see or error messages

### Actual results

[crash_Tue_Feb_20_22_28_11_0.log](https://github.com/SoftFever/OrcaSlicer/files/14354428/crash_Tue_Feb_20_22_28_11_0.log)
[crash_Tue_Feb_20_22_28_52_0.log](https://github.com/SoftFever/OrcaSlicer/files/14354430/crash_Tue_Feb_20_22_28_52_0.log)

### Expected results

Should allow login to Bambu Labs cloud service without having to type user/pass.

### Project file & Debug log uploads

[logs_2-20-2024.zip](https://github.com/SoftFever/OrcaSlicer/files/14354467/logs_2-20-2024.zip)

### Checklist of files to include

- [X] Log file
- [ ] Project file

jjarmoc, 2024-03-02:
I came looking for this issue as it's also impacting me. Since I don't have creds for Bambu/Makerworld but rather use OAuth, this leaves me unable to login to Orcaslicer.

If any further information is needed I'd be happy to provide what I can.

MaxRink, 2024-03-22:
+1, this is still happening in 2.0.0-beta

DrewFitz, 2024-03-26:
Also getting a similar crash on macOS when trying to sign in via Sign in with Apple. The thread crashes trying to start an HTTP server to receive the login response.

```
Process:               OrcaSlicer [15704]
Path:                  /Applications/OrcaSlicer.app/Contents/MacOS/OrcaSlicer
Identifier:            com.softfever3d.orca-slicer
Version:               1.9.1 (???)
Code Type:             ARM-64 (Native)
OS Version:            macOS 14.4.1 (23E224)
Exception Type:        EXC_CRASH (SIGABRT)
Exception Codes:       0x0000000000000000, 0x0000000000000000

Termination Reason:    Namespace SIGNAL, Code 6 Abort trap: 6
Terminating Process:   OrcaSlicer [15704]

Application Specific Information:
abort() called

~~~ lots of threads ~~~

Thread 17 Crashed:
0   libsystem_kernel.dylib        	       0x19d72aa60 __pthread_kill + 8
1   libsystem_pthread.dylib       	       0x19d762c20 pthread_kill + 288
2   libsystem_c.dylib             	       0x19d66fa20 abort + 180
3   libc++abi.dylib               	       0x19d719d30 abort_message + 132
4   libc++abi.dylib               	       0x19d709fcc demangling_terminate_handler() + 320
5   libobjc.A.dylib               	       0x19d3a81e0 _objc_terminate() + 160
6   libc++abi.dylib               	       0x19d7190f4 std::__terminate(void (*)()) + 16
7   libc++abi.dylib               	       0x19d71c348 __cxxabiv1::failed_throw(__cxxabiv1::__cxa_exception*) + 88
8   libc++abi.dylib               	       0x19d71c28c __cxa_throw + 308
9   OrcaSlicer                    	       0x103123988 void boost::throw_exception<boost::system::system_error>(boost::system::system_error const&, boost::source_location const&) + 72
10  OrcaSlicer                    	       0x10312ef90 boost::asio::basic_socket_acceptor<boost::asio::ip::tcp, boost::asio::any_io_executor>::basic_socket_acceptor<boost::asio::io_context>(boost::asio::io_context&, boost::asio::ip::basic_endpoint<boost::asio::ip::tcp> const&, bool, boost::asio::constraint<is_convertible<boost::asio::io_context&, boost::asio::execution_context&>::value, int>::type) + 1284
11  OrcaSlicer                    	       0x10312d94c boost::detail::thread_data<Slic3r::GUI::HttpServer::start()::$_2>::run() + 96
12  OrcaSlicer                    	       0x103729a2c boost::(anonymous namespace)::thread_proxy(void*) + 176
13  libsystem_pthread.dylib       	       0x19d762f94 _pthread_start + 136
14  libsystem_pthread.dylib       	       0x19d75dd34 thread_start + 8
```

jjarmoc, 2024-04-01:
Update: For me, 2.0.0-beta resolves the issue when using Google login. Unable to test with AppleID.

elofgren, 2024-04-14:
I have been having this issue using AppleID.

```
-------------------------------------
Translated Report (Full Report Below)
-------------------------------------

Process:               OrcaSlicer [61687]
Path:                  /Applications/OrcaSlicer.app/Contents/MacOS/OrcaSlicer
Identifier:            com.softfever3d.orca-slicer
Version:               2.0.0 ()
Code Type:             ARM-64 (Native)
Parent Process:        launchd [1]
User ID:               501

Date/Time:             2024-04-14 00:26:53.0419 -0700
OS Version:            macOS 14.4.1 (23E224)
Report Version:        12
Anonymous UUID:        CCD1FD77-9404-E95E-2F54-086F9557E4E5

Sleep/Wake UUID:       98D24D82-3EFD-474C-951A-13793F7D8968

Time Awake Since Boot: 590000 seconds
Time Since Wake:       1663 seconds

System Integrity Protection: enabled

Crashed Thread:        19  http_server

Exception Type:        EXC_CRASH (SIGABRT)
Exception Codes:       0x0000000000000000, 0x0000000000000000

Termination Reason:    Namespace SIGNAL, Code 6 Abort trap: 6
Terminating Process:   OrcaSlicer [61687]

Application Specific Information:
abort() called

Thread 19 Crashed:: http_server
0   libsystem_kernel.dylib        	       0x18cf3aa60 __pthread_kill + 8
1   libsystem_pthread.dylib       	       0x18cf72c20 pthread_kill + 288
2   libsystem_c.dylib             	       0x18ce7fa20 abort + 180
3   libc++abi.dylib               	       0x18cf29d30 abort_message + 132
4   libc++abi.dylib               	       0x18cf19fcc demangling_terminate_handler() + 320
5   libobjc.A.dylib               	       0x18cbb81e0 _objc_terminate() + 160
6   libc++abi.dylib               	       0x18cf290f4 std::__terminate(void (*)()) + 16
7   libc++abi.dylib               	       0x18cf2c348 __cxxabiv1::failed_throw(__cxxabiv1::__cxa_exception*) + 88
8   libc++abi.dylib               	       0x18cf2c28c __cxa_throw + 308
9   OrcaSlicer                    	       0x1060ec4a4 void boost::throw_exception<boost::system::system_error>(boost::system::system_error const&, boost::source_location const&) + 72
10  OrcaSlicer                    	       0x1060f01f4 boost::asio::basic_socket_acceptor<boost::asio::ip::tcp, boost::asio::any_io_executor>::basic_socket_acceptor<boost::asio::io_context>(boost::asio::io_context&, boost::asio::ip::basic_endpoint<boost::asio::ip::tcp> const&, bool, boost::asio::constraint<is_convertible<boost::asio::io_context&, boost::asio::execution_context&>::value, int>::type) + 1284
11  OrcaSlicer                    	       0x1060eeba4 boost::detail::thread_data<Slic3r::GUI::HttpServer::start()::$_4>::run() + 124
12  OrcaSlicer                    	       0x1064a1fb0 boost::(anonymous namespace)::thread_proxy(void*) + 176
13  libsystem_pthread.dylib       	       0x18cf72f94 _pthread_start + 136
14  libsystem_pthread.dylib       	       0x18cf6dd34 thread_start + 8

Thread 19 crashed with ARM Thread State (64-bit):
    x0: 0x0000000000000000   x1: 0x0000000000000000   x2: 0x0000000000000000   x3: 0x0000000000000000
    x4: 0x00005ffeb334ef7a   x5: 0x0000000000000002   x6: 0x0000000000000020   x7: 0x0000000000000b90
    x8: 0x92257006231bd72b   x9: 0x9225700530d3a72b  x10: 0x0000000000001600  x11: 0x00000000a140902e
   x12: 0x00000000000007fb  x13: 0x00000000000007fd  x14: 0x00000000a160982d  x15: 0x000000000000002d
   x16: 0x0000000000000148  x17: 0x00000001fefeac30  x18: 0x0000000000000000  x19: 0x0000000000000006
   x20: 0x0000000313c87000  x21: 0x000000000002db73  x22: 0x0000000313c870e0  x23: 0x0000000000000001
   x24: 0x00000001089eaa98  x25: 0x0000000000000000  x26: 0x00000001089eab1e  x27: 0xb2ab117a257edfd0
   x28: 0x0000000108c51000   fp: 0x0000000313c86cf0   lr: 0x000000018cf72c20
    sp: 0x0000000313c86cd0   pc: 0x000000018cf3aa60 cpsr: 0x40001000
   far: 0x0000000000000000  esr: 0x56000080  Address size fault

```

netenken, 2024-06-15:
I have the same issue with google login

closed 2024-09-21 by github-actions[bot]

theosib, 2025-01-02:
This is still a bug in 2.3.0. I suggest reopening it and marking it as a blocker.

lohmatij, 2025-04-04:
Same for me on 2.3.0 

Can't login because of this bug

wkethman, 2025-07-09:
Agree - same issue