OrcaSlicer/OrcaSlicer #4196: Seam Painting - Sphere Bug

TinyMito, 2024-02-23:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

1.9.1

### Operating System (OS)

macOS

### OS Version

macOS 14

### Additional system information

MacBook Pro 14 M1 Pro

### Printer

Bambu Lab P1S

### How to reproduce

Open any model or large model.
1. Select Object.
2. Click Seam Painting
3. Click Sphere mode

### Actual results

Sphere paint indicator is off the model.

https://github.com/SoftFever/OrcaSlicer/assets/75095713/5b499291-1cb8-4894-9251-d7ccf5a08885

### Expected results

Should be on the model.

### Project file & Debug log uploads

[Long+dobble+box.3mf.zip](https://github.com/SoftFever/OrcaSlicer/files/14380948/Long%2Bdobble%2Bbox.3mf.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

TinyMito, 2024-03-23:
Refer to #4652

TinyMito, 2024-03-23:
Hi, @Noisyfox I got to reopen this bug report. I just tested out the 2.0.0 RC. The sphere bug is still there on macOS 14, M1 Pro.

Video here:

https://github.com/SoftFever/OrcaSlicer/assets/75095713/36566a9e-dbd2-4b8a-9ac3-eb8a96199e6e

Noisyfox, 2024-03-23:
[image]
Works fine here. Make sure you downloaded the latest master/nightly build.

TinyMito, 2024-03-23:
I have reinstalled the Orca 2.0.0 RC 2nd time, the bug is gone now. Thank you!

closed 2024-03-23 by TinyMito