OrcaSlicer/OrcaSlicer #4198: Print existing gCode to Bambu printers

tlhintoq, 2024-02-23:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

You can either load, SLICE and print a job.
Or you can load, slice and SEND the job to the printer SD card then get up and start it from the machine.

It would be really nice if you could load, saved gCode and print it remotely.  Shop management where all the work files are on the computer.  The gCode is proven and saved.  No additional time required to reslice. No mistakes from different settings.

### Which printers will be beneficial to this feature?

Marlin

### Describe the solution you'd like

I should like to be able to load a gCode that was sliced with Orca, then hit [Print].
I'm thinking it should be exactly the same as slicing to generate the gcode then printing, only without the additional slicing time, and without the possibility of a mistake in settings since the last time that job was run.

[image]
[image]

### Describe alternatives you've considered

Seems pretty straight forward.  Enable the [Print] button upon successful gCode preview instead of on successful slicing completion.

### Additional context

Any gcode file produced by Orca should do

samwiseg0, 2024-02-24:
This can be achieved today and this enhancement is not need. Bambu printers use 3mf files that contain gcode files in them. All you would need to do is export as a 3mf after you slice.

Export the sliced 3mf:
[image]

After that file is created open the file with orca and you will be presented with the "Print Plate" button as soon as the file is loaded. 

[image]

tlhintoq, 2024-02-24:
That's awesome - Thank you!
When I first read it was yet another use of the .3mf I wondered how tough keeping track of all the different ways that file type is used, would be.
But... As it is formatted with the same variables from the solution just like gCode output, its pretty clear this is a print file versus the project file.  Terrific news!
[image]

closed 2024-02-24 by tlhintoq

samwiseg0, 2024-02-24:
Yes. You can always check the contents of  3rd file by changing it to .zip

You will then be able to see the contents of the 3mf by unzipping the file.