OrcaSlicer/OrcaSlicer #4199: Exported .json profiles does not work in the CLI mode

grepson, 2024-02-23:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

1.9.1

### Operating System (OS)

Linux, macOS

### OS Version

MacOS Ventura 13.2.1

### Additional system information

Orcaslicer is running in the docker container with base of node:20-bullseye-slim

### Printer

Anycubic Vyper with Stealthburner

### How to reproduce

1. Export settings for process / printer in .json format
2. Load them in the CLI mode via `--load-settings` parameter

### Actual results

Running command within docker:

```
docker run orcaslicer-node:latest \
 --debug 5 \
 --slice 0 --info \
 --load-settings "./profiles/orcaslicer/printer/vyper_sb.json" \
 --outputdir ./output/orcaslicer \
 ./stl/3DBenchy.stl
```

The file is always invalid, the CLI cannot read it, spitting out an error: "unknown config type  of file ./profiles/orcaslicer/printer/vyper_sb.json"

### Expected results

CLI should process the imported .json file and slice the model.

### Project file & Debug log uploads

Docker output log:
```
[2024-02-23 04:48:26.260785] [0x00007f1eae6fac80] [trace]   Initializing StaticPrintConfigs
[2024-02-23 04:48:26.481369] [0x00007f1eae6fac80] [warning] cli mode, Current OrcaSlicer Version 01.08.02.56
[2024-02-23 04:48:26.483643] [0x00007f1eae6fac80] [info]    Will start to read model file now, file count :1

[2024-02-23 04:48:26.483793] [0x00007f1eae6fac80] [info]    allow_multicolor_oneplate 1, allow_rotations 1 skip_modified_gcodes 0 avoid_extrusion_cali_region 0 loaded_filament_ids size 0, clone_objects size 0
[2024-02-23 04:48:26.483803] [0x00007f1eae6fac80] [info]    plate_to_slice=0, normative_check=1, use_first_fila_as_default=0
[2024-02-23 04:48:26.485440] [0x00007f1eae6fac80] [info]    read model file:./stl/3DBenchy.stl

[2024-02-23 04:48:26.652015] [0x00007f1eae6fac80] [debug]   TriangleMesh::repair() started
[2024-02-23 04:48:26.811803] [0x00007f1eae6fac80] [debug]   TriangleMesh::repair() finished
[2024-02-23 04:48:27.073592] [0x00007f1eae6fac80] [info]    object 3DBenchy.stl, id :5, from stl or other 3mf

[2024-02-23 04:48:27.073998] [0x00007f1eae6fac80] [info]    run:before load settings, file count=1

[2024-02-23 04:48:27.075765] [0x00007f1eae6fac80] [info]    operator():load setting file ./profiles/orcaslicer/printer/vyper_sb.json, with rule 1

operator(): unknown config type  of file ./profiles/orcaslicer/printer/vyper_sb.json in load-settings[2024-02-23 04:48:27.088292] [0x00007f1eae6fac80] [info]    record_exit_reson:433, saved config to ./output/orcaslicer/result.json

run found error, return -5, exit...
[2024-02-23 04:48:27.088353] [0x00007f1eae6fac80] [info]    cli_callback_mgr_t::stop enter.
[2024-02-23 04:48:27.088358] [0x00007f1eae6fac80] [info]    cli_callback_mgr_t::stop not started before, return directly.
```

result.json:

```
{
    "error_string": "The input preset file is invalid and can not be parsed.",
    "export_time": 0,
    "plate_index": 0,
    "prepare_time": 0,
    "return_code": -5
}
```

### Checklist of files to include

- [X] Log file
- [X] Project file

### Anything else?

Also the cli documentation is very limited, I don't even know if my settings are right to slice a file correctly, since there is no testing done for that. I'm unable to slice any file with saving it, I'm getting a json with result if something is wrong and nothing if the file is sliced.

grepson, 2024-02-25:
I was able to run the .json's exported from bambustudio docs: https://github.com/bambulab/BambuStudio/wiki/Command-Line-Usage. However this still results in no export being done.

Command: 
```
SLICER=orcaslicer

 docker run -v $PWD/output/${SLICER}:/${SLICER}/output/${SLICER} \
 -v $PWD/profiles:/${SLICER}/profiles \
 -v $PWD/stl:/${SLICER}/stl \
 --rm --device /dev/fuse --cap-add SYS_ADMIN \
 ${SLICER}-node:latest \
 --debug 6 \
 --outputdir ./output/orcaslicer \
 --slice 0 \
 --load-settings "profiles/orcaslicer/printer/machine.json;profiles/orcaslicer/print/process.json" \
 --load-filaments "profiles/orcaslicer/filament/filament.json" \
 ./stl/3DBenchy.stl
```

Output:
[slicing_result.json](https://github.com/SoftFever/OrcaSlicer/files/14394845/slicing_result.json)

eduncan911, 2024-05-31:
not stale.  keep open!

jdh313, 2024-08-31:
Not stale!

closed 2024-12-09 by github-actions[bot]