OrcaSlicer/OrcaSlicer #4202: The software crashes when I print on my P1S

RafBois, 2024-02-23:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

1.9.1

### Operating System (OS)

macOS

### OS Version

14.2.1 (23C71)

### Additional system information

Apple M1 Max

### Printer

P1S

### How to reproduce

The software crashes when I print on my P1S 

[image]
[image]

### Actual results

[image]

### Expected results

Unable to print with Orcaslicer 1.9.1. This worked perfectly before installing version 1.9.1.
I used it every day

### Project file & Debug log uploads

[Archive.zip](https://github.com/SoftFever/OrcaSlicer/files/14383579/Archive.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Anything else?

[image]
[image]

Jer406, 2024-02-28:
im having them same issue on windows

git-rz, 2024-03-03:
Same issue on windows, but the printer is Voron Trident running latest Mainsail / Klipper.

closed 2024-06-21 by github-actions[bot]