OrcaSlicer/OrcaSlicer #4210: Please Add Device: Anycubic Kobra 2 Plus

aldaw, 2024-02-23:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

I have downloaded the nightly 2.0 and the Anycubic Kobra 2 Plus is missing in the list.
You can download and the settings from their official homepage for prusiaslicer.
https://store.anycubic.com/pages/firmware-software
https://drive.google.com/file/d/1D-iqxPlP9TRq5NXVWtQRBSRd3YL2Ip79/view?usp=drive_link

### Which printers will be beneficial to this feature?

Klipper

### Describe the solution you'd like

The Anycubc Kobra 2 Plus is listed.

aldaw, 2024-02-23:
Here is a script to convert it in .json for orcaslicer.
https://github.com/theophile/SuperSlicer_to_Orca_scripts

cropse, 2024-05-06:
I've same feature request, thanks.

cropse, 2024-05-06:
OK, I tried and it ends out not good https://github.com/theophile/SuperSlicer_to_Orca_scripts/issues/31

White3rabbit, 2024-05-15:
I have the same problem.

NnamOnye, 2024-08-24:
Please integrate Anycubic Kobra 2 Plus into the orca slicer

cropse, 2024-08-27:
@NnamOnye Try this and read the comment: https://www.printables.com/model/788326-kobra-2-pro-profiles-for-orcaslicer

PR #7082 (2024-10-22): Add Anycubic Kobra 2 Plus, Kobra 2 Max, Kobra 2 Pro, Kobra 2 Neo, & Kobra 3
Thanks to @soosp for providing me the files from Anycubic's Slicer!

Closes #4210, closes  #3332, closes #6200, closes #5048, and closes #6269
commit 09afb390 (2024-10-22): Add Anycubic Kobra 2 Plus, Kobra 2 Max, Kobra 2 Pro, Kobra 2 Neo, & Kobra 3 (#7082)

Thanks to @soosp for providing me the files from Anycubic's Slicer!

Closes #4210, closes #3332, closes #6200, closes #5048, and closes #6269
closed 2024-10-22 by SoftFever