OrcaSlicer/OrcaSlicer #4215: Arc fitting wall overflow problem

olcayoren, 2024-02-24:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

1.9.1

### Operating System (OS)

macOS

### OS Version

14.2.1 Sonoma

### Additional system information

Air M1 Arm

### Printer

Elegoo Neptune 4 Pro

### How to reproduce

1- Add Cylinder
2- On arc fitting

### Actual results

**Arc fitting on:**
[image]
[image]

**Arc fitting off:**
[image]
[image]

### Expected results

None

### Project file & Debug log uploads

[Cylinder.3mf.zip](https://github.com/SoftFever/OrcaSlicer/files/14392657/Cylinder.3mf.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

igiannakas, 2024-02-25:
Your resolution is set to 0... Set it back to default and try again

olcayoren, 2024-02-26:
> Your resolution is set to 0... Set it back to default and try again

With default values, there are still deformities on the walls

**Arc fitting on:**

[image]
[image]

**Arc fitting off:**

[image]
[image]

igiannakas, 2024-02-26:
With klipper arc fitting should be turned off as klipper does not create arc movements - it breaks them down to segments (lines) defined by the resolution in the printer.cfg.

if you print the cylinder on a Bambu or marlin printer you they will come out artefacts free.

olcayoren, 2024-02-26:
> With klipper arc fitting should be turned off as klipper does not create arc movements - it breaks them down to segments (lines) defined by the resolution in the printer.cfg.
> 
> if you print the cylinder on a Bambu or marlin printer you they will come out artefacts free.

I didn't know thatklipper doesn't support g2 and g3 commands. When I examined it in detail, I confirmed that it was not supported. Thank you for your information. 👏

closed 2024-02-27 by olcayoren