OrcaSlicer/OrcaSlicer #4216: Arc fitting not working using ender profile but works if I modify the bambu profile for the ender.

CodeMasterCody3D, 2024-02-24:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

1.9

### Operating System (OS)

Windows

### OS Version

10

### Additional system information

Ryzen 3600 16fb ram rx 5700xt

### Printer

Ender 3 Pro 

### How to reproduce

If I have 2 instances opened up of Orca, one with bambu profile and the other ender profile I copied all settings over to the bambu for the ender and vice versa trying to get the arc fitting to work like the modified bambu profile that works on the ender 3 with arc fitting. Arc fitting does not work like it does on bambu profile while on ender profile despite any settings changes like slowing down for overhangs.

### Actual results

The ender profile does not incorporate the arc fitting properly on the chimney area of the 3d benchy (circular object) 

### Expected results

I should see G2 commands for arc circular motion on gcode for the chimney area on the 3d benchy but I do not.

### Project file & Debug log uploads

[](url)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Anything else?

Simple fix possibly for now is to add function of octoprint server add on for the bambu profile So i can quickly upload.

hidara2000, 2024-03-14:
My following comment may be related. 

On some profiles I can check the arc fitting option but for some profiles, it is greyed out

https://github.com/SoftFever/OrcaSlicer/assets/15170494/65077982-f477-44cc-9189-6392e0baf0cc

dJOS1475, 2024-04-16:
I can confirm this Arc fitting bug is real - if I select a Voron default profile it’s not greyed out:

[image]

However, on profiles I created from the Voron Standard profile on earlier versions of OS, it's greyed out:

[image]

CanaryTours, 2024-04-29:
Same here, I can't find what is not letting me activate the Arc fitting, did you find any solution?

PR #5352 (2024-05-18): Updated arc fitting pop up to clarify that this option should not be enabled for Klipper machines
# Description

Klipper does not natively support arc commands at the motion controller level. Hence enabling arc fitting on Klipper is detrimental to print quality as line segments are converted to arcs in the slicer and back to line segments in Klipper.

The revised pop up wording clarifies that this option is not recommended for Klipper printers. 

Fixes #5297 
Fixes #4216
closed 2024-05-18 by SoftFever