OrcaSlicer/OrcaSlicer #4220: Calibration tool for Klipper Square Corner Velocity

chrisheib, 2024-02-25:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

There is no tool and very little information online about tuning Klipper Square Corner Velocity and the results in printing quality.

### Which printers will be beneficial to this feature?

Klipper

### Describe the solution you'd like

Not really sure where I'm going with this issue, but I wrote a small klipper macro to allow me to tune Square Corner Velocity, the Klipper equivalent of Jerk, from Orca.

This isn't really a final version, just some thoughts that I would love to be searchable online, therefor the ticket. Where would be most appropriate to post this?

Klipper Macro:

```
[gcode_macro SET_SQV]
gcode:
  {% set multiplier = params.MULTIPLIER|default(0.01)|float %}
  {% set layer = params.LAYER|default(1)|float %}
  {% set sqv = multiplier * layer %}
  RESPOND TYPE=echo MSG='{"set sqv: multi: %f, layer: %f, final sqv: %f" % (multiplier, layer, sqv) }'
  SET_DISPLAY_TEXT MSG='{"S%.1f M%.2f L%.0f " % (sqv, multiplier, layer) }'
  SET_VELOCITY_LIMIT SQUARE_CORNER_VELOCITY={ sqv }
```

Orca Layer Change GCODE:

``` 
SET_SQV MULTIPLIER=0.2 LAYER=[layer_num]
``` 

Use a [small calibration geometry file like this.](https://github.com/SoftFever/OrcaSlicer/files/14394821/SCV-Calib.zip)

Scale the model height according to the layer amount you want to test and set a reasonable multiplier in the layer change gcode.

I recommend to draw a seam on the flat face in the back to prevent Orca from putting it into a circle: 
[image]

Slice with 2 or 3 walls, no top and bottom layers, add mouse ear brims, and let it rip. 

To find your best value, keep an eye on the print, your printer and the screen. The current SCV is shown as 'S7.5' for example. You can also use calipers like with most other calibration tower prints.

My findings: 

The biggest gains in quality as well as speed are found in the low numbers, especially in circles around 2 to 5 mm radius, as those are rendered as many small lines with quite sharp angles. Because the extrusion is really staggered with many pauses in the curves at SCV of under 1 I could see a noticeable different, shiny material look, which faded at higher SCVs.

I don't think I see any difference in printer behavior or layer speed above 10, so I think I will set mine to 10 and call it a day. I found some posts online saying 'your printer will jump off of your table at values higher than 10', which disappointingly didn't happen in my experiment. For now I'm only driving my printer at 250 mm/s and 1500mm²/s until I get my accelerometer and can set up input shaping. 

I guess CSV tuning should be done as a final step, after all other calibrations have been done, especially input shaping, extruder step calibration, flow ratio, and pressure advance.

[image]

[image]

[image]

I think setting a starting value in the macro could be nice, as well as printing the first layer with a safe CSV value like 3, because printing the mouse ears with 0 CSV is so darn painfully slow.

chrisheib, 2024-02-25:
@dstulken Gotcha, great input, thanks for the feedback.

Not sure if I have the patience right now to model something with varying amplitudes, but this should be a lot better already, right?

[image]

chrisheib, 2024-02-25:
I like this one a lot more, but results have been the same. I'll try again with the diagonal edge being 90° to stress test x-y-transitions.

[image]

chrisheib, 2024-02-25:
Still no perceivable difference in quality, even at 15 SCV. Layers get printed quite a bit faster, so i guess thats a win. 

I'm not entirely sure what this tells me yet. I think it shows my printer has a lot more headroom to increase speeds, even without input shaping, which is... unexpected, it being a quite cheap Geetech A20M thats modded to smithereens. Even without properly tensioned belts and a noobishly wonky direct extruder setup. 

I'll try scaling up the model to see if longer faces -> higher speeds lead to different results.

tkunchick, 2024-02-29:
I think you need longer straight runs in the model to let the printer get to full speed before trying to slow for the turn.

SoftFever, 2024-04-23:
Nice one!
I like the first test model as it allows the printhead to reach max speed before the turn.

> The biggest gains in quality as well as speed are found in the low numbers, especially in circles around 2 to 5 mm radius, as those are rendered as many small lines with quite sharp angles. Because [...]

This is consistent with my observation too. I have investigated cases where users report that curves are not smooth. The root cause is they accidentally set jerk to 1, which creates a lot of artifacts when printing curved surfaces.

I will do some test when I got time

Anzii, 2024-09-21:
I believe that the main reason why your printer doesn't jump around is the fact that your general acceleration is too low.
Try the same sawtooth model with 10k accel

8Chachie6, 2024-11-11:
> ### Is there an existing issue for this feature request?
> * [x]  I have searched the existing issues
> 
> ### Is your feature request related to a problem?
> There is no tool and very little information online about tuning Klipper Square Corner Velocity and the results in printing quality.
> 
> ### Which printers will be beneficial to this feature?
> Klipper
> 
> ### Describe the solution you'd like
> Not really sure where I'm going with this issue, but I wrote a small klipper macro to allow me to tune Square Corner Velocity, the Klipper equivalent of Jerk, from Orca.
> 
> It prints a reference object and increases SCV by a certain amount each layer, like a lot of the other calibration towers.
> 
> This isn't really a final version, just some thoughts that I would love to be searchable online, therefor the ticket. Where would be most appropriate to post this?
> 
> Klipper Macro:
> 
> ```
> [gcode_macro SET_SQV]
> gcode:
>   {% set multiplier = params.MULTIPLIER|default(0.01)|float %}
>   {% set layer = params.LAYER|default(1)|float %}
>   {% set sqv = multiplier * layer %}
>   {% if layer <= 1 %}
>     {% set sqv = 2 %}
>   {% endif %}
>   RESPOND TYPE=echo MSG='{"set sqv: multi: %f, layer: %f, final sqv: %f" % (multiplier, layer, sqv) }'
>   SET_DISPLAY_TEXT MSG='{"S%.1f M%.2f L%.0f " % (sqv, multiplier, layer) }'
>   SET_VELOCITY_LIMIT SQUARE_CORNER_VELOCITY={ sqv }
> ```
> 
> Orca Layer Change GCODE:
> 
> ```
> SET_SQV MULTIPLIER=0.2 LAYER=[layer_num]
> ```
> 
> Use a [small calibration geometry file like this.](https://github.com/SoftFever/OrcaSlicer/files/14396797/SCV-V2.zip)
> 
> [Old model](https://github.com/SoftFever/OrcaSlicer/files/14394821/SCV-Calib.zip)
> 
> Scale the model height according to the layer amount you want to test and set a reasonable multiplier in the layer change gcode.
> 
> I recommend to draw a seam on the flat face in the back to prevent Orca from putting it into a circle: ![image](https://private-user-images.githubusercontent.com/6112968/307559001-9811fc7e-d930-4e61-90fc-11cc8f1dfb99.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzEzNjE4NzMsIm5iZiI6MTczMTM2MTU3MywicGF0aCI6Ii82MTEyOTY4LzMwNzU1OTAwMS05ODExZmM3ZS1kOTMwLTRlNjEtOTBmYy0xMWNjOGYxZGZiOTkucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MTExMSUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDExMTFUMjE0NjEzWiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9YzI0NjFlZGRkNmRhMGNkNjkwMTNhNzNjNzIyNDUyNTYwOTI3ZGNkZjM2NjZmMWNmM2ZjMDRmMTcxODI4OTc2YyZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QifQ.sJ3FZAujVppXxz8Wbryzy-yC3qhzgo2NbVsg-18rCDE)
> 
> Slice with 2 or 3 walls, no top and bottom layers, add mouse ear brims, and let it rip.
> 
> To find your best value, keep an eye on the print, your printer and the screen. The current SCV is shown as 'S7.5' for example. You can also use calipers like with most other calibration tower prints.
> 
> My findings:
> 
> The biggest gains in quality as well as speed are found in the low numbers, especially in circles around 2 to 5 mm radius, as those are rendered as many small lines with quite sharp angles. Because the extrusion is really staggered with many pauses in the curves at SCV of under 1 I could see a noticeable different, shiny material look, which faded at higher SCVs.
> 
> I don't think I see any difference in printer behavior or layer speed above 10, so I think I will set mine to 10 and call it a day. I found some posts online saying 'your printer will jump off of your table at values higher than 10', which disappointingly didn't happen in my experiment. For now I'm only driving my printer at 250 mm/s and 1500mm²/s until I get my accelerometer and can set up input shaping.
> 
> I guess CSV tuning should be done as a final step, after all other calibrations have been done, especially input shaping, extruder step calibration, flow ratio, and pressure advance.
> 
> ![image](https://private-user-images.githubusercontent.com/6112968/307558817-c1d7c99c-d142-4638-9bba-44e6eabf5000.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzEzNjE4NzMsIm5iZiI6MTczMTM2MTU3MywicGF0aCI6Ii82MTEyOTY4LzMwNzU1ODgxNy1jMWQ3Yzk5Yy1kMTQyLTQ2MzgtOWJiYS00NGU2ZWFiZjUwMDAucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MTExMSUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDExMTFUMjE0NjEzWiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9YjUyYWQ2MjI1ZTU3ZTJkMzNhYTMyY2U2MjRkZTUzZDI1YzI1Y2YwNjZjNjg4NTJjZWNmZTZjM2RiNWQyMTRlMiZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QifQ.OQieWiW9AEMEv7xPptQMAvzL0REQNNeCMVFDyCvaL_o)
> 
> ![image](https://private-user-images.githubusercontent.com/6112968/307558825-eda97486-88de-4258-8a4f-b715ea29ff94.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzEzNjE4NzMsIm5iZiI6MTczMTM2MTU3MywicGF0aCI6Ii82MTEyOTY4LzMwNzU1ODgyNS1lZGE5NzQ4Ni04OGRlLTQyNTgtOGE0Zi1iNzE1ZWEyOWZmOTQucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MTExMSUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDExMTFUMjE0NjEzWiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9MzYyNzJmZTZjMzE2MmNmYzBhNjhiNDlhODBjMzhjMjI3OGViMzkyZDNjMmU2YjYyNTA [... cut ...] DQ2NmEzNWZmMWNiMSZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QifQ.NB-n3g_Zj8Pbo09vJibdrypnUTYTPtmM_cV0DwvZO3w)
> 
> I think setting a starting value in the macro could be nice, ~as well as printing the first layer with a safe CSV value like 3, because printing the mouse ears with 0 CSV is so darn painfully slow.~

yea hi this is confusing .... what do i do with the macro and what do you do with the line put in the slicer after what numbers are going where like it dont explain that ?? what am i doing here ??

chrisheib, 2024-11-12:
@8Chachie6 haha, this is quite advanced stuff for me, so apologies if it's not super clear 😄 

Basically I put the macro into my clipper config so I can use the SET_SQV command at any point from gcode. 

In the orca slicer 'Custom gcode on layer change'-field I then entered the SET_SQV call with a few parameters so the corner velocity is increased a bit after each layer. 

Thats basically all the magic there is to this. Once thats done you can print just any file and watch the printer speed up the higher it goes. 

My test files where just quickly thrown together in onshape. They made for interesting tests when printed with 0 bottom, 0 top and no infill, so basically just 2 or 3 walls. Vase mode might be fun too.

chrisheib, 2025-02-27:
Interesting, didn't know that one was so generic. Great to know!

As the macro is already done, you might as well use that with some custom stl. If the towers geometry is sufficient for your tests, it's a great option tho!

freakydude, 2025-02-28:
> Interesting, didn't know that one was so generic. Great to know!
> 
> As the macro is already done, you might as well use that with some custom stl. If the towers geometry is sufficient for your tests, it's a great option tho!

One more info - the TUNING_TOWER command has nothing todo with the klipper `square_tower.stl`. Its just a generic command that calls a given gcode command with values depending on Z.  See klipper doc [TUNING_TOWER](https://www.klipper3d.org/G-Codes.html?h=tuning_tower#tuning_tower_1)

So you can use it with any stl and with any command or macro you like

ianalexis, 2025-10-20:
Did this here #10962 
I would like that you could check it and test it.