OrcaSlicer/OrcaSlicer #4237: Slicing bug multiple materials

hutje, 2024-02-26:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

1.9.1

### Operating System (OS)

Windows

### OS Version

Windows 10

### Printer

Bambu Lab X1C

### How to reproduce

Click Slice with provided 3mf file

### Actual results

Filament 1 (orange) is being used instead of filament 2 (yellow) during the first top shell layer.

### Expected results

Filament 2 should be used for the whole yellow object.

### Project file & Debug log uploads

[Multi material bug.zip](https://github.com/SoftFever/OrcaSlicer/files/14406429/Multi.material.bug.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

tkunchick, 2024-02-29:
Might want to describe how you painted the model for the multi color. 

Seems like a painting error rather than a slicer error. Though does slice just fine on MAC Orca. I don't have a windows machine to test on.

[image]

hutje, 2024-02-29:
> Might want to describe how you painted the model for the multi color.
> 
> Seems like a painting error rather than a slicer error. Though does slice just fine on MAC Orca. I don't have a windows machine to test on.
> 
> <img alt="Screenshot 2024-02-29 at 11 54 36 AM" width="1236" src="https://private-user-images.githubusercontent.com/45150915/309007693-607e8e27-a111-496e-82cd-6c550cc15b24.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDkyMjc3ODUsIm5iZiI6MTcwOTIyNzQ4NSwicGF0aCI6Ii80NTE1MDkxNS8zMDkwMDc2OTMtNjA3ZThlMjctYTExMS00OTZlLTgyY2QtNmM1NTBjYzE1YjI0LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAyMjklMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMjI5VDE3MjQ0NVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTA0ODQyZGFjY2U4ODI5Y2FjNzI5NmI2MGVlNGZlNjNhNDU1Y2Q5MzUxMGEwNzBiODAxNjRmNGY3MGE2ZTZmNzYmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.TvWjH0IGvAy7Ujwi3MR54-2JP66XuKq4obtwdJ7Raxs">
Not painted. Just a multi part object. This problem does not occur in newest version of Bambu Studio.

closed 2024-06-07 by github-actions[bot]