OrcaSlicer/OrcaSlicer #4239: Add custom text message to display screen for Bambu X1C during pause

nathan-cowley, 2024-02-26:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

We utilize the pause function frequently for inserting magnets in 3D-printed parts. The magnets almost always need to be inserted in a specific orientation (we color one side black to dictate the poles). It would be very convenient to allow for a custom message to display on the screen (or be added onto the default pause message) that in our case says something like "3 magnets black down".

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

In the slicer when you add a pause there would be an option to display additional text at pause where you could type in a text box.

3DJupp, 2024-02-28:
Would love to see that.
"Insert NFC Tag", "change filament".. etc.

kylek29, 2024-02-28:
I don't think this is possible as the printer firmware handles it -- maybe contact the X1Plus firmware guys and see if it's something they can implement ( https://www.x1plus.net/ )? In the past I've manually added the G-codes that would normally push a message to a screen and the X1C just ignores them. Bambu seems to have a custom function call (you can see it in some of the printer-specific G-code) which hardcoded messages.

nathan-cowley, 2024-03-01:
Bummer but that makes sense.

closed 2024-03-01 by nathan-cowley