OrcaSlicer/OrcaSlicer #4252: Can't create new filament presets with imported printer presets

Apotrox, 2024-02-28:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

1.9.1

### Operating System (OS)

Windows

### OS Version

Windows 11

### Additional system information

Ryzen 9 5900x, RTX 3080, 32GB 3200mhz RAM

### Printer

Anycubic Kobra Neo

### How to reproduce

1. Import custom printer profile 
2. Try to create a filament preset for that printer

### Actual results

Filament Presets created under the system printer preset will not show up for the custom printer presets. 

### Expected results

Being able to create filament presets under the custom printer preset (or preferably without being tied to any printer at all!)

### Project file & Debug log uploads

[debug_Wed_Feb_28_17_25_57_16764.log.zip](https://github.com/SoftFever/OrcaSlicer/files/14437526/debug_Wed_Feb_28_17_25_57_16764.log.zip)
[new.zip](https://github.com/SoftFever/OrcaSlicer/files/14437543/new.zip)
Note that the project file doesnt contain a model (but hopefully the presets)
[Printer presets.zip](https://github.com/SoftFever/OrcaSlicer/files/14437602/Printer.presets.zip)
Here are the printer presets

### Checklist of files to include

- [X] Log file
- [X] Project file

### Anything else?

[image]
When trying to create the filament with a existing preset (which doesnt exist)
[image]

[image]
Doesnt appear under my custom printer presets
[image]
Only for the generic klipper preset

Apotrox, 2024-02-28:
Maybe i just can't use OrcaSlicer correctly but i'd also like to point out that adding my 3 existing filament presets was already incredibly cumbersome to do due to the exact same issue.

GitHag42, 2024-03-04:
I wanted to open the very same topic:
Can't create Custom Filaments for my custom (imported) printer preset.

I see a usecase for printer related filament profiles: e.g. different retract overwrites for different printers (bowden or direct extruder) OR when real temperatures significantly differ on different printers for the same set value OR different fan speeds are needed due to different setups.
But it would also be nice to assign a filament profile to several printers at once if such a differentiation is not needed.

By the way, where do you get all the filament information from (density, shrinkage, softening temp, ...)?

Apotrox, 2024-03-04:
@GitHag42 from the manufacturers website usually (see [here](https://www.elegoo.com/en-de/products/elegoo-pla-3d-printer-filament-1-75mm-colored-1kg) for example). Everything i don't know i just leave as is.

karaktaka, 2024-06-04:
It seems like this problem still exists in the current Beta. In Version 2.0.0 it worked sometimes but I had multiple "Printers" as choice (I suspect it was always the same as it showed the exact same name multiple times), but now in Beta 2.1.0 I don't see any printers anymore when trying to add a custom filament.
[image]

closed 2024-09-10 by github-actions[bot]