OrcaSlicer/OrcaSlicer #4261: User Profiles not available in Second OrcaSlicer Instance

spikeon, 2024-02-29:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

1.9.1

### Operating System (OS)

Windows

### OS Version

Windows 11

### Printer

Elegoo Neptune 4 Max

### How to reproduce

Open Orca Slicer
Open Orca Slicer Again
Try to select customized printer profile on second instance

### Actual results

User Profiles section is missing

### Expected results

User Profiles are there and selectable.  Maybe to prevent conflicts, profile selected in first instance is disabled. 

### Project file & Debug log uploads

This is not a crash.  

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Anything else?

I have multiple printers with very different bed sizes and I want to prepare the next plate for both of them at the same time.  If I could use multiple instances of Orca Slicer I could do this, but when I try to do so I lose the ability to use my profiles.

tkunchick, 2024-02-29:
Works fine on MAC. I do this all the time.

Now if you change a setting on one, the other instance will not see the change until it is restarted.  Which makes sense it only reads the configs at start up.

spikeon, 2024-03-01:
This is on Windows. 

[image]

Here's a screenshot with both instances open.  On the left is my first instance with my N4PRO.4 profile selected.  On the right is the second instance.   Note that none of the User Profiles are present.

LJKraus, 2024-03-05:
Can't reproduce problem on Windows 11, OrcaSlicer 1.9.1.  I had up to five instances open, all Orca User printer profiles were available in all, plus two additional instances of Bambu Studio for supported printers.

LOiPFiT, 2024-04-27:
I have this issue too. On first boot, Orca shows me only system profiles. I need to restart it and only after that user profiles are available.

caste1200, 2024-04-28:
I have the same issue. Its not about having multiple instances of orca, its just that 1 out of 2 orca startups, it loads different "profiles(?)", I open the slicer once and its all good, I close it and open it again and its like it all has gone back to default from when I installed Orca. Somehow orca slicer has 2 different default profiles?

winbard, 2024-05-25:
Hello,

I have the same problem with version 2. (Windows 10 Pro)
First start of orca slicer i have only the printer which i choose on after the installation. Close and open again I have my printers and profiles. When I wait too long to start twice it is like the first start.
After I install orca i import my profiles from version before for information.

caste1200, 2024-05-25:
How can I check the profiles the software will load at startup? How can I overwrite wich profiles it will load? It must be conflicting with different ini files or something??

caste1200, 2024-06-13:
> How can I check the profiles the software will load at startup? How can I overwrite wich profiles it will load? It must be conflicting with different ini files or something??

Someone?

closed 2024-09-20 by github-actions[bot]