OrcaSlicer/OrcaSlicer #4262: Can't Slice file to send to A1 Mini

skl111, 2024-02-29:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.0.0-DEV

### Operating System (OS)

Windows

### OS Version

11

### Printer

Bambu Labs A1 Mini

### How to reproduce

Attempt to slice a file using the A1 Mini Profile

### Actual results

It produces this error
[image]

### Expected results

Slice should complete without errors

### Project file & Debug log uploads

[debug_Wed_Feb_28_23_12_03_254688.log.zip](https://github.com/SoftFever/OrcaSlicer/files/14442716/debug_Wed_Feb_28_23_12_03_254688.log.zip)
[Cube.zip](https://github.com/SoftFever/OrcaSlicer/files/14442718/Cube.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

SoftFever, 2024-03-03:
Thank you for reporting this.
Fixed now.
The fix will be included in the nightly release once the auto build complete

closed 2024-03-03 by SoftFever

skl111, 2024-03-03:
Welcome, and thank you!

skl111, 2024-03-03:
I can confirm the build fixed it, just sliced and diced it into an A1 Mini

commit 27b8c349 (2024-03-03): fix in_head_wrap_detect_zone issue
fixes #4262