OrcaSlicer/OrcaSlicer #4265: gcode to anycubic --- Damaged file, cannot view details

sysnxw, 2024-02-29:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

1.9.1

### Operating System (OS)

Windows

### OS Version

win10

### Printer

kobra2pro

### How to reproduce

Must appear.

### Actual results

G files uploaded to cloud-universe.anycubic.com, When viewing or printing, the web page prompts "Damaged file, cannot view details"

### Expected results

You can view it through the web, and the app prints Gcode.

### Project file & Debug log uploads

“”Damaged file, cannot view details“”

### Checklist of files to include

- [ ] Log file
- [ ] Project file

nyonor2, 2024-03-25:
so?

A-lexLuthor, 2024-04-08:
Hello, same issue here. I search every where but can't find a solution

Frengo92, 2024-05-16:
Any news about the problem?

closed 2024-08-22 by github-actions[bot]