OrcaSlicer/OrcaSlicer #4266: One infill direction on all layers

yellowicz, 2024-02-29:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

1.9.1

### Operating System (OS)

macOS

### OS Version

Monterey

### Printer

Voron Trident

### How to reproduce

Set Top shell layers to 0
Set Bottom shell layers to 0
Set Sparse infill density to 100%
Set Sparse infill pattern to Aligned Rectilinear
Set Infill direction to 0°

### Actual results

Since version 1.9 this has changed and each layer is printed at a right angle to the previous one. It doesn't matter what angle I set in the Infill direction field.
[image]

### Expected results

In the previous version of Orca, 1.8, if I wanted to get a clear print from transparent PETG, I could eliminate the top and bottom layers and set sparse infill to Aligned Rectilinear 100%. With the Infill direction setting of 0°, each infill was printed in the same direction.
https://wiki.bambulab.com/en/knowledge-sharing/transparent-petg
[image]

### Project file & Debug log uploads

[H2LOT - PCTG.3mf.zip](https://github.com/SoftFever/OrcaSlicer/files/14446403/H2LOT.-.PCTG.3mf.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

Ataraxia-Mechanica, 2024-03-08:
At 100%, the slicer converts all sparse infill to solid infill. Changing the infill percentage to 99.9999% stops this from happening.

jeor112, 2024-03-11:
Thanks Ataraxia-Mechanica, I had the same issue with OrcaSlicer and it helped to reduce the infill from 100% to 99.999/%
It doesn't seem to be an issue in the latest OrcaSlicer.

GuzziRaz, 2024-05-05:
There is one issue still, but it can be worked around: If you set top/bottom layers to zero, the "Internal solid infill pattern" option is greyed out so you can't select Aligned rectilinear. Yet a 100% value for sparse infill will use that setting. However, the setting IS honored, so if you set it to Aligned rectilienar and only then set top/bottom layers to zero, you wont get the cross hatching. Either that, or set sparse infill to 99.99% which is more of a bodge fix.

So the bug in Orca now AFAICS is the greying-out of "Internal solid infill pattern", it should never be greyed out.

On the other hand, Bambu Studio also greys it out BUT it will also not use that pattern setting for 100% sparse infill. It will still use the pattern for sparse infill. I think that is even more logical.

LYXOfficial, 2024-05-28:
Maybe it is not a good change, add a option to choose which direction I want to use is a better way.

jasonconstant33, 2024-06-10:
@GuzziRaz changing the number of top&bottom layers back to 1, then changing the solid infill pattern to aligned rectilinear worked for me in latest version too. Unfortunately I did not notice this issue of my layers not being aligned until a few hours into the print, so I will leave the part with rectilinear infill, not the end of the world.

closed 2024-09-16 by github-actions[bot]