OrcaSlicer/OrcaSlicer #4270: Fan Control Fail - Starts fans at 100% instead of filament setting.

tkunchick, 2024-02-29:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

1.9.1

### Operating System (OS)

macOS

### OS Version

Sonoma 14.3.1 (23D60)

### Printer

K1 Max

### How to reproduce

**Set fan speeds in filament profile**

[image]

**slice multi layer file**
[image]

**check fan speed at different levels in preview and gcode**
[image]
[image]

### Actual results

Fans start at 100% regardless of filament profile settings.

**Corrects it self once it hits something that changes fan speed such as over hang or support and then runs at correct filament setting.**
[image]

### Expected results

Starting at indicated level, fans speed set to setting as indicated in filament profile.

### Project file & Debug log uploads

No log

### Checklist of files to include

- [ ] Log file
- [ ] Project file

closed 2024-06-07 by github-actions[bot]

Asiak3, 2024-12-20:
Same issue V2.2.0

Chevan94, 2025-05-13:
It's the same issue in all the recent versions of Orca.
The M106 S255 seems to get inserted right after the filament change gcode, resulting in turning the fan on full blast, instead of the setting dictated by the filament. 

I tried a lot of things, Bambu Studio doesn't do that. 

Has anyone managed to work around this somehow?

I'm on BambuLab A1.