OrcaSlicer/OrcaSlicer #4276: Device Tab (WebView process) used CPU

snkku, 2024-03-01:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.0-dev

### Operating System (OS)

Windows

### OS Version

Windows 11

### Printer

FlyingBear Ghost 6

### How to reproduce

When application running, WebView process used 10-16% CPU, even if application minimized (not used)
Klipper(Mainsail) inside

### Actual results

In any state WebView process using 10-15% CPU, that's discharging battery and heating notebook CPU.

### Expected results

Stop/Disable/Sleep activity in Device(WebView) component, while this tab is not active! Or enable feature to disable this tab at all.

### Project file & Debug log uploads

no file

### Checklist of files to include

- [ ] Log file
- [ ] Project file

Ditchbuster, 2024-03-14:
I can confirm having this problem too. 
Windows 11 Pro - brand new build and OS install. 
2.0-dev
Connecting to Klipper Mainsail

Ditchbuster, 2024-03-14:
So as I said, new install, new OS. Had previously been accessing mainsail via web (brave) on windows 10. Accessing using brave on windows 11 pro also causes CPU spike but when you change view off the tab, its stops its high CPU (unlike orca which keeps using high CPU).

Ditchbuster, 2024-03-15:
@snkku I was able to workaround the problem by going into the mainsail config/.theme folder (click gears and show hidden files) into the .css file and delete the top section between the comments for the background animation. I don't know if this is a problem with animation with mainsail or if the way it was setup was very unoptimized.

besser, 2024-04-01:
I having this problem too. My setup:
- Windows 11 Pro 23H2, old OS install
- OrcaSlicer 2.0
- Printer running Klipper and Fluidd

closed 2024-07-09 by github-actions[bot]

vorrin, 2024-11-19:
Hello, I also always had this problem, it's not major but I'd much prefer to have no CPU sink when the app is minimized or simply when the device tab is not selected, if anyone's looking at doing a bit of polishing, this is one of the very few issues I could point out with this wonderful program.