OrcaSlicer/OrcaSlicer #4285: Only one wall at top surfaces issue

streetgt, 2024-03-01:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

1.9.1

### Operating System (OS)

Windows

### OS Version

Windows 10

### Additional system information

Not revelant

### Printer

Not revelant

### How to reproduce

With Only one wall at top is creating some kind of layer artifacts, if disabled works great.

With Only one wall at top surfaces - Enabled
[image]

Only one wall at top surfaces - Disabled
[image]

SuperSlicer
[image]

File provided to reproduce.
[Q-59-ZL-59.zip](https://github.com/SoftFever/OrcaSlicer/files/14466858/Q-59-ZL-59.zip)

### Actual results

Top surface contains "artifacts"

### Expected results

Surface should not contain "artifacts"

### Project file & Debug log uploads

No logs needed

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Anything else?

No

streetgt, 2024-03-01:
Bambu Slicer:
[image]

Orca Slicer:
[image]

igiannakas, 2024-03-02:
Have you tried the latest nightly? There was a fix pushed for the one wall top layer.

streetgt, 2024-03-02:
Still not fixed. This is an annoying bug that should be fixed.

[image]

streetgt, 2024-03-02:
@SoftFever can we have a fix on this instead of pushing new features?

igiannakas, 2024-03-02:
Check super slicer and Bambu slicer if they are using Arachne for slicing with one wall, where it works. I think they are not showing this bug because of use of classic slicing.

there is an issue with Arachne plus this feature on some rare occasions. In this instance the workaround is to set classic slicing for that part.

tiagoacardoso, 2024-03-02:
No, changing between archne and classic doesn’t fix the issue.

igiannakas, 2024-03-02:
[image]
set one wall threshold to 100%.

SoftFever, 2024-03-03:
For models with text embossment, it's recommended to set the threshold to 0%.

It is mentioned in the tooltip
[image]

closed 2024-03-05 by streetgt