OrcaSlicer/OrcaSlicer #4297: Error in custom gcode: cannot apply not operator

rbrian, 2024-03-02:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.0.0 dev, build triggered by merging scarf seam pull request 

### Operating System (OS)

Windows

### OS Version

11

### Printer

Bambu A1 mini

### How to reproduce

- load a file
- set "scarf joint seam (beta)" to other than None
- Slice plate

### Actual results

I get an error message: 

Failed to generate gcode for invalid custom G-code.
timelapse_gcode Parsing error at line 36: Cannot apply a not operator.
  {if !in_head_wrap_detect_zone}
       ^
Please check the custom G-code or use the default custom G-code.

### Expected results

Successful slice

### Project file & Debug log uploads

[Pole support scarf error.zip](https://github.com/SoftFever/OrcaSlicer/files/14470452/Pole.support.scarf.error.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

rbrian, 2024-03-02:
Happened on multiple files.

rbrian, 2024-03-02:
Also appears in a couple-day-old OrcaSlicer_Windows_PR3839_portable

Noisyfox, 2024-03-03:
Same as #4262 which has been fixed now in the latest nightly build (once the current build is completed).

closed 2024-03-03 by Noisyfox