OrcaSlicer/OrcaSlicer #4298: Still issues with vase mode / spiral mode after "full re-write of spiral vase" (1.9.1)

m50434, 2024-03-02:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

1.9.1

### Operating System (OS)

Windows

### OS Version

Windows 11

### Printer

BambuLab P1S

### How to reproduce

Hi,
I downloaded the last version of orca slicer - esecially because there is the "[Fill re-write of spiral case](https://github.com/SoftFever/OrcaSlicer/pull/3091)" included.

I didnt get it exactly but as I understand the new "Smoothing option" is only for particular objects:
"Because some vase mode prints are designed for on one layer being one shape then the next layer being a completely different shape and rely on clever bridging to make it printable. If you smooth those then you end up completely breaking the intention of the design."

### Actual results

Therefore my object WITH smoothing it looks like this:
[image]

If I turn it off it looks quite much better, but there are still some litte irregular movements:

[image]

I guess its caused by the round faces the object has.

### Expected results

No irregular movements on some (random?) places.

### Project file & Debug log uploads

[test.zip](https://github.com/SoftFever/OrcaSlicer/files/14470555/test.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

m50434, 2024-06-02:
Hi @andrewboktor 
is there something new? :-)

bzwart77, 2024-07-04:
Hello, I have been doing a lot of testing with Vase mode prints Using Orca Slicer 2.1.1 and my general observations are that the Smooth spiral works miracles on models that are more organic or rounded curves.  It will completly remove the existing vase mode seam many of us are familar with.  That said, when enabling on a a model that has hard edges it seems to make a mess out of the mesh.  The following photo is the visible seam when smooth spiral is off.  
[image]

The following photo is with Smooth Spiral on and the default Max XY Smoothing of %200
[image]
As you can see the mesh is a complete mess.  

At this point I use it for any model that does not have hard edges as it works wonders removing the seam however on many other models its tearing up the mesh so I have to turn it off on a case by case basis.  Please let me know if you want any sample models to further help testing this as I am seeing this behavior on quite a few models.

m50434, 2024-10-03:
Still interested in getting the issue fixed

bzwart77, 2024-10-03:
Same here, if there is any way I can help please let me know.

yazz007, 2024-12-10:
I've also noticed strange head movement during print without spiral smoothing, most of the layer print as expected and some of them are a bit shifted outside just on small part of layer (i.e 1cm), like in pictures above.

The second problem is that with retraction set to on I see a Z-seam which is not in the preview. Cura prints perfectly fine in spiral mode. I don't understand also why I cannot set more walls in bottom surfaces - at that point spiral movement is not yet active, right? Spiral should start at layer above last bottom.

I love OrcaSlicer but those issues are quite important to me, and for the spiral prints I have to switch to Cura which doesn't have cool top/bottom patterns like archimedean chords which perfectly match spiral mode.

j0e57, 2024-12-26:
I'm using normal straight vase mode (not shifting X and Y) but manage to me it print 2wall like using Orca. But when I add base on it it act strangely .. after base complete it start 1 wall mode for about 3 layer then it start 2 wall mode after automaticcally with thin wall below. I see it in slicer and sure not want to print it.. still not yet to try cura.
Expected behaviour is after base complete it just start 2wall vase mode like there is no base, maybe add 1 layer act as base for next layer in 2wall. May be I include photo later if needed

j0e57, 2024-12-26:
When trying to photo what I mean I see 3 layer wall is wrong so I try to add bottom layer to 6 (set 3 before) and the 7th layer correctly start the vase mode with 2wall, when I set 5 bottom it left 1 false layer. It seems the model need 6 layer for bottom but when set 3 it not added to 6 automatically to fullfill model requirement.. seems its only on vase mode so still valid for this issue but maybe different fixed needed, thanks

andrewboktor, 2025-01-14:
Apologies for being MIA here (that's why the feature can be turned off!).
I managed to replicate the result shown in the screenshots but unusure why this happens. I will *try* to find some time to take a look.

andrewboktor, 2025-02-02:
Alright, I found some time today to dig in. I'm pretty confident I found two bugs that impacted this model. Here's what it looks like after those bugs have been fixed.

[image]

There is still some very minor artifacts; I suspect those might be caused by rounding errors.

andrewboktor, 2025-02-03:
After spending a whole weekend on this. I concluded that there is likely only one bug, not two. My second bug fix seemed to improve things but after going through the math a billion times, it is incorrect math. So I took it out. As of now, I don't have idea on how to improve this in any significant way.

j0e57, 2025-02-11:
@andrewboktor so it is fixed on nightly builts? will try because I now facing quite same problem with my double wall design that use twisting..weird jump on Z axis on random layer so orca not produce not smooth wall, trying on cura it smooth but cura use different path (change direction) while I prefer Orca that continue vase mode in one direction tough seems like there is like 1 point bridged but actually its raise Z on that spot :)

closed 2025-05-20 by github-actions[bot]