OrcaSlicer/OrcaSlicer #4299: "Access violation" with this STL repeated 4 times

PPCM, 2024-03-02:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

1.9.1

### Operating System (OS)

Windows

### OS Version

Windows 11 Pro

### Printer

Voron 2.4R2

### How to reproduce

1. Take the STL file in attachment
2. copy it 4 times
3. Mirror 2 items along Y Axis
4. Arrange all objects with the following option : Auto rotate
5. Slice

### Actual results

Orca craches with "Access violation" error

If during the Arrange step, the option "Auto rotate" is not activated, Orca doesn't crash

### Expected results

Slices without error

### Project file & Debug log uploads

Log file: 
[debug_Sat_Mar_02_20_28_10_17296.log.0.zip](https://github.com/SoftFever/OrcaSlicer/files/14470552/debug_Sat_Mar_02_20_28_10_17296.log.0.zip)

Project file and STL file:
[Couvercle.zip](https://github.com/SoftFever/OrcaSlicer/files/14470559/Couvercle.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

### Anything else?

I made lot of tries, the issue comes from the element with a rotation of 90°
If all elements have no rotation, there is no issue

foreachthing, 2024-03-02:
I get the same error (Windows 11 Pro, OS 1.9.1).
But if I slice (your 3mf) with my Profile, it slices anyway I like...

Noisyfox, 2024-03-03:
What you can do as a workaround is to manually rotate the left most model by 90 degs then rotate back. Meanwhile I'm still investigating the root cause of this issue.

PPCM, 2024-03-03:
Thanks guys for your help
@Noisyfox  In fact, this is why I made lot of test, it is to find a workaround :-) I don't permit the rotation of elements during the arrangement, and the slicing works fine, but I loose space on the plate
If you need more information, feel free to ask

PPCM, 2024-03-03:
An information, if it can help
With a rotation of 45°, the slicing works fine

closed 2024-06-09 by github-actions[bot]