OrcaSlicer/OrcaSlicer #4302: Better support for modifiers, convert any object to modifier

pmarkiewicz, 2024-03-02:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

I have few model when modifier is well positioned inside step file (2 objects), after load and split will be very useful  to mark one as modifier.

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

Right click on model and chenge to modifier

### Describe alternatives you've considered

None, current solution with add modifier from file is problematic due to problem with good positioning

closed 2024-06-09 by github-actions[bot]

DBT85, 2024-11-10:
I would like to add a vote for this. Being able to change a part to be a modifier would be very useful.

kalmdown, 2025-01-06:
Would like to load a model with multiple parts so some can be turned into modifiers that can be parametrically built in place in the CAD program.

DBT85, 2025-01-06:
> Would like to load a model with multiple parts so some can be turned into modifiers that can be parametrically built in place in the CAD program.

I'm not sure if it changed since I commended in November or if I've just looked at it in a different way, but when importing a step file that contains multiple parts, you can right click on one and select "change type" and then choose from Part, Negative Part, Modifier, Support Blocker or Support Enforcer.

This only works BEFORE you "split to objects". It cannot for some reason be done after being split.

So if, for example, you make a Fusion file with a sphere component, with a cylinder component running through it, you can export that whole model as a step file, import to Orca and then chose which of those bodies within that one "object" is a modifier or negative part.

kalmdown, 2025-01-06:
That is a solution for what I wanted to accomplish. 

The user just needs to know it's the expected tool path when things like Modifiers need to be precisely placed.

I can still see value in turning any object's type, but that's not a priority for me.