OrcaSlicer/OrcaSlicer #4308: Segmentation fault (core dumped)

goombah88, 2024-03-03:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

1.9.1 and 1.9.0

### Operating System (OS)

Linux

### OS Version

EndeavourOS (arch linux)

### Printer

Bambu Lab P1S

### How to reproduce

Double click to run the AppImage or drag it to a terminal and hit ENTER.

### Actual results

Orca slicer launches, shows the home screen, I can't interact with the app at all, then after a few seconds crashes with a Segmentation fault (core dumped). It happens on 1.9.0 and 1.9.1 both installed from the AUR and using the AppImages. I last used it a week ago and it worked fine. I also have the HarmonyOS and NanumGothic fonts installed already.

Full output:
[2024-03-02 19:48:00.414064] [0x0000760ed5096480] [trace]   Initializing StaticPrintConfigs
add font of HarmonyOS_Sans_SC_Bold returns 1
add font of HarmonyOS_Sans_SC_Regular returns 1
add font of NanumGothic-Regular returns 1
add font of NanumGothic-Bold returns 1
loading network module at /home/goombah/.config/OrcaSlicer/plugins/libbambu_networking.so
after dlopen, network_module is 0x605dcfc55850
BambuTunnel::GetMsg(1)
[2024-03-02 19:48:04.606441 -05:00] [I] [T 8823]: TunnelMqtt::reconnect: result: 0

(orca-slicer:8799): GLib-GObject-CRITICAL **: 19:48:04.715: invalid cast from 'wxPizza' to 'GtkCellLayout'

(orca-slicer:8799): Gtk-CRITICAL **: 19:48:04.715: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:8799): GLib-GObject-CRITICAL **: 19:48:04.716: invalid cast from 'wxPizza' to 'GtkCellLayout'

(orca-slicer:8799): Gtk-CRITICAL **: 19:48:04.716: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:8799): GLib-GObject-CRITICAL **: 19:48:04.716: invalid cast from 'wxPizza' to 'GtkCellLayout'

(orca-slicer:8799): Gtk-CRITICAL **: 19:48:04.716: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:8799): GLib-GObject-CRITICAL **: 19:48:04.717: invalid cast from 'wxPizza' to 'GtkCellLayout'

(orca-slicer:8799): Gtk-CRITICAL **: 19:48:04.717: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:8799): GLib-GObject-CRITICAL **: 19:48:04.717: invalid cast from 'wxPizza' to 'GtkCellLayout'

(orca-slicer:8799): Gtk-CRITICAL **: 19:48:04.717: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:8799): GLib-GObject-CRITICAL **: 19:48:04.718: invalid cast from 'wxPizza' to 'GtkCellLayout'

(orca-slicer:8799): Gtk-CRITICAL **: 19:48:04.718: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:8799): GLib-GObject-CRITICAL **: 19:48:04.718: invalid cast from 'wxPizza' to 'GtkCellLayout'

(orca-slicer:8799): Gtk-CRITICAL **: 19:48:04.718: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:8799): GLib-GObject-CRITICAL **: 19:48:04.719: invalid cast from 'wxPizza' to 'GtkCellLayout'

(orca-slicer:8799): Gtk-CRITICAL **: 19:48:04.719: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed
[... 298 lines cut ...]
(orca-slicer:8799): GLib-GObject-CRITICAL **: 19:48:06.198: invalid cast from 'wxPizza' to 'GtkCellLayout'

(orca-slicer:8799): Gtk-CRITICAL **: 19:48:06.198: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:8799): GLib-GObject-CRITICAL **: 19:48:06.198: invalid cast from 'wxPizza' to 'GtkCellLayout'

(orca-slicer:8799): Gtk-CRITICAL **: 19:48:06.198: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:8799): GLib-GObject-CRITICAL **: 19:48:06.198: invalid cast from 'wxPizza' to 'GtkCellLayout'

(orca-slicer:8799): Gtk-CRITICAL **: 19:48:06.198: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed
Segmentation fault (core dumped)

### Expected results

App should load and be usable.

### Project file & Debug log uploads

[log.zip](https://github.com/SoftFever/OrcaSlicer/files/14471235/log.zip)

### Checklist of files to include

- [X] Log file
- [ ] Project file

goombah88, 2024-03-03:
Sometime it gives me truncated output:

2024-03-02 19:47:46.263153] [0x0000774662af7480] [trace]   Initializing StaticPrintConfigs
add font of HarmonyOS_Sans_SC_Bold returns 1
add font of HarmonyOS_Sans_SC_Regular returns 1
add font of NanumGothic-Regular returns 1
add font of NanumGothic-Bold returns 1
Unhandled unknown exception; terminating the application.

jasperfect, 2024-03-04:
same issue, don't know if something relates to Wayland

fredrikasberg, 2024-03-04:
Seems to be a problem with the network plugin.
If you remove the plugin the application starts fine but instantly downloads the plugin and that in turn makes the application segfault again.

[orcaslicer_gdb_output.txt](https://github.com/SoftFever/OrcaSlicer/files/14481538/orcaslicer_gdb_output.txt)

goombah88, 2024-03-04:
I was able to use the appimage for 1.8.1 to print something. Every time I
open it though it re-downloads the network plugin.

fredrikasberg, 2024-03-05:
> I was able to use the appimage for 1.8.1 to print something. Every time I open it though it re-downloads the network plugin.
> […](#)

https://github.com/SoftFever/OrcaSlicer/issues/4178#issuecomment-1966641383

Use this network plugin and manually copy it to `~/.config/OrcaSlicer/plugins` before running OrcaSlicer (you can use 1.9.1).

goombah88, 2024-03-07:
> > I was able to use the appimage for 1.8.1 to print something. Every time I open it though it re-downloads the network plugin.
> > […](#)
> 
> [#4178 (comment)](https://github.com/SoftFever/OrcaSlicer/issues/4178#issuecomment-1966641383)
> 
> Use this network plugin and manually copy it to `~/.config/OrcaSlicer/plugins` before running OrcaSlicer (you can use 1.9.1).

Worked like a charm! I'm back on 1.9.1 again. Thanks for the tip buddy.

closed 2024-06-14 by github-actions[bot]

TheDevMinerTV, 2024-09-04:
For me OrcaSlicer crashes like this (after installing via `yay -S orca-slicer-bin`):
```
$ orca-slicer 
[2024-09-04 20:13:55.239519] [0x0000724258ae4b80] [trace]   Initializing StaticPrintConfigs
add font of HarmonyOS_Sans_SC_Bold returns 1
add font of HarmonyOS_Sans_SC_Regular returns 1
add font of NanumGothic-Regular returns 1
add font of NanumGothic-Bold returns 1
Segmentation fault (core dumped)
```

adlerweb, 2024-09-06:
@TheDevMinerTV In this case t's probably a bug related to font rendering. Try to remove harmonyos-sans: `pacman -Rdd ttf-harmonyos-sans`

sTiKyt, 2025-06-24:
> Orca bot: this issue is stale because it has been open for 90 days with no activity.

Bruh

BlathanAevon, 2025-07-14:
> > Orca bot: this issue is stale because it has been open for 90 days with no activity.
> 
> Bruh, this is a joke, right? The issue is still there, why'd you close it with a bot, it crashes for people, it was properly reported
> 
> What, are we supposed to daily chat on an issue just so it doesn't get closed?

spent like 5 hours trying to solve segmentation fault, nothing, even building from source doesn't help...

sTiKyt, 2025-08-26:
Just in case someone's looking for a fresh issue:
https://github.com/SoftFever/OrcaSlicer/issues/10524