OrcaSlicer/OrcaSlicer #4309: OrcaSlicer crashes on opening / crashes after trying to synchronize

Modusmundi, 2024-03-03:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

1.9.0 & 1.9.1

### Operating System (OS)

Linux

### OS Version

Garuda Linux

### Additional system information

12th Gen Intel i5-12600K (16) @ 4.900GHz, NVIDIA GeForce RTX 3070 Ti, 32GB RAM, xfce 4.18, 6.6.19-1-lts kernel.

### Printer

Bambu X1C

### How to reproduce

Orca has taken to crashing when I open it.  I'm on Linux, and I cleared out my ~/.config/OrcaSlicer/ folder.  This lets me reconfigure fine.  Reinstalling the network driver seems to go fine.  When I attempt to log back into my Bambu account for the printer, it starts up a bar that says "synchronizing" and then crashes.  

### Actual results

Any subsequent attempts to load the program crash shortly after loading.  

### Expected results

I expect to be able to open the application and function normally.  Issue does not occur in v1.8.1.

### Project file & Debug log uploads

[03022024_log.zip](https://github.com/SoftFever/OrcaSlicer/files/14471291/03022024_log.zip)

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Anything else?

The logs don't appear to show any critical errors, which is super weird.  More than happy to help try and generate more logs or do anything here- If I run it at the command line it generates some fun GTK errors, but those are par for the course and are in 1.8 as well.

ddt3, 2024-03-03:
This appears to be a duplicate of #4178

Modusmundi, 2024-03-03:
@ddt3 It appears as such, I'll give the workaround mentioned in the thread a go.  I must have missed that when I did a quick search.  Thanks and apologies.  Will close after I repro results from #4178.

Modusmundi, 2024-03-04:
@ddt3 Can confirm this resolves.  Will await patched network driver I guess while using the old version.  Thanks a ton.

closed 2024-03-04 by Modusmundi