OrcaSlicer/OrcaSlicer #4324: Add "Speed values" to Settings overwrite by filament

Z18Aplha, 2024-03-03:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

I use filament which needs to be printed very slowly to achieve a certain look (BioFusion by Extruder). But it can be printed very fast for unvisible traces like infill or inner walls, so the volumetric capabilities are set to a medium/high value.

So when i print with this filament (or with this filament in a multi color print) i must configure the object speed settings evereytime. It is very annoying because the setting is mainly filament dependant.

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

It would be nice to add a new section to the filament "setting overrides" which addresses the "outer wall speed". 

### Describe alternatives you've considered

1) A new process profile for the filament
- not working because it is only usable, when printing a single color print
2) Changing the speed in the local object modifiers
- works, but is annoying and leads to mistakes for bigger prints with many parts

mtmello2003, 2024-03-13:
I miss this feature a lot. 

Could be very usefull

erdo100, 2024-03-13:
Yes, I just had the same issue also with Extrudr Inca Gold. I solved it by setting the  max volumetric flow for the whole filament. I would also prefer to set the speeds in the filamant override settings.

Z18Aplha, 2024-03-14:
@erdo100 the problem with this fix is, that also infill and inner walls are printed that slow. 
for the BioFusion Filament only the outer walls need to be printed slowly.

I found out that 15-20mm/s lead to great metallic effect. This is far to slow to set the volumetric flow for the whole print.

erdo100, 2024-03-14:
@Z18Aplha yes, you are right, I just had accepted it to print slow for each line type.

closed 2024-06-21 by github-actions[bot]

rickbeec, 2024-09-17:
I would like to reactivate this discussion, as I have some of the same issues as the OP. My workaround is to have many Presets which I save inside the projects. This unfortunately doesn't help the next time I use the same filament. I find that the Max Vol Speed is a pretty blunt instrument in a very, very tweakable slicer. I would love to see a copy of the "Other Layers Speed" inside the Filament Settings>Setting Overrides.

KarleHeinz, 2024-11-05:
I have the same problem. Is there meanwhile a solution? 
This feature would be very nice because you don't have to change the speeds manually when you print for example TPU instead of PLA.
Thanks

LifeBlast, 2024-11-11:
I have the same issue, I want to override speed limits per filament.

Deanxr5, 2025-02-07:
It would be great to override speed settings per filament.

Nandox7, 2025-04-17:
Same here, for some filaments I have to lower at least the first layer speed for it to adhere well.
It would make more sense to have it as a override that is linked to the specific filament (s) than having to create a specific separate profile.

andreamerello, 2025-06-12:
Yes, and I guess that issues like this could potentially proliferate for anything (now it's extrusion rate smoothing for TPU for me). 

IMO It would be nice to implement some way to override ideally anything in the filament profiles (speeds included) i.e. you can "add" any setting to the filament profile override list.

JFingerle, 2025-10-26:
> Why would anyone limit the settings available in overrides?

It's a software limitation I guess, it's not by choice.

QQuark, 2025-11-19:
Also would be interested in this feature for a similar use case: I noticed that printing an ASA first layer really slow (20-30mm/s) improves drastically how good it sticks to the bed, but I don't want this on PETG/PLA/etc.

Maybe add a scripting area allowing to just write parameter=value in the filament settings ? This would solve the edge case needs but avoid having to manage hundreds of additional widgets and without complexifying the UI.

bartagergely, 2026-01-02:
Here is my +1 for this.

I have some filaments for example which can be printed with the "regular" speed, but the first layer must be slowed down. 

I have two options now:

- Set the non-regular speed rates by hand before each print
- Under the Process section to have all the variations for each layer height X every filament I use

Speed override fields in the filament settings would be a huge help for this.

0x4bit, 2026-02-20:
This. Is. Needed.
With the lot of toolchanging printers becoming available, it bothers a lot that you are basically able to use them as multi-color printers but not as multi-material ones.