OrcaSlicer/OrcaSlicer #4326: Flow Dynamics Calibration Silliness

DARKFiB3R, 2024-03-03:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

1.9.0

### Operating System (OS)

Windows

### OS Version

OS: Windows 10

### Additional system information

CPU: i7-4790K 4.4GHz
RAM: 16GB
GFX: GTX 970

### Printer

Bambu Lab P1S

### How to reproduce

Attempt to start Flow Dynamics Calibration

### Actual results

Can't continue due to invalid default values

Suggested values also do not work.

Wording is not consistent, i.e. "From K Value" = "Start Value", etc.

[image]

[image]

[image]

### Expected results

Default values should be usable.

Sugested values should work.

Wording should be consistent (everywhere in the application tbh)

### Project file & Debug log uploads

N/A

### Checklist of files to include

- [X] Log file
- [X] Project file

### Anything else?

Thanks

Yorshka-Vermilion, 2024-03-23:
i've encountered this one in BS too. In BS you would need to replace "." with "," on numbers. Don't know if this will work in Orca too, but its basiclly the same system so if you are in hurry it is worth give it a try.

frankieorabona, 2024-03-27:
Even inserting the comma always gives the same problem:

[image]

frankieorabona, 2024-03-27:
I have the same problem. 
Even inserting the comma always gives the same problem:

[image]

By the way, in Bambu Studio with the same values it works regularly

patridge, 2024-04-04:
Have this issue on OrcaSlicer 2.0.0. Didn't have any luck swapping the decimal separator to commas, though. I don't know enough about the test to risk trying other values, but everything from the error-suggested to the default values presents an error.

nmcandrews, 2024-05-28:
I am seeing this issue for the first time today as well. 
Same exact parameters for both instances.
Starts fine in Bambu Studio, but gives an error in Orca Slicer. 
Orca Slicer will actually CRASH to desktop, when trying to run a pattern as the calibration shape, rather than a line, while citing an error of invalid operation. But Bambu Studio is more than happy to do it.

Jinro87, 2024-05-30:
Exactly the same happens on my Orca Slicer too. Default values show this invalid values error. 
It seems like it accepts "corrected" values but the whole Slicer crashes right after.
This happened across few older versions of the Slicer already.
I think the last version where flow dynamics calibration worked for me was 1.8.x.x.

closed 2024-09-05 by github-actions[bot]