OrcaSlicer/OrcaSlicer #4328: Toolbar icons blurry

yw4z, 2024-03-04:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.0

### Operating System (OS)

Windows

### OS Version

10 LTSC

### Printer

Creality K1

### How to reproduce

Changing window width aligns toolbar to perfect center that makes start point of toolbar as float number instead of int. that causes blurry images

### Actual results

here is comparison of perfect fit vs slightly shifted blurry icons

[image]

i changed window width few times to catch perfect fit

### Expected results

i guess current placement of toolbar if sidebar on left
start_point_of_toolbar = ((window_width - sidebar_width ) / 2 )- (toolbar_width / 2)

number should be convert to "int" after calculation to fix that otherwise window tries to render toolbar start with decimal numbers and that causes blurry images

### Project file & Debug log uploads

-

### Checklist of files to include

- [ ] Log file
- [ ] Project file

discip, 2024-03-08:
@Noisyfox
Could you have a look please? 😊

yw4z, 2024-03-08:
also i found toolbar > emboss > text alignment and style buttons blurry. these could be related with this issue

Noisyfox, 2024-03-08:
Sure I'll take a look soon. BTW thanks for your icon rework, that's amazing~

yw4z, 2024-03-08:
Thanks for your efforts on orca, way better than other ones. just wanted to help on project to make it better

yw4z, 2024-03-13:
@Noisyfox 
i dont know how to build to test but i guess this line should be int instead float
https://github.com/SoftFever/OrcaSlicer/blob/eb5a1bc7135462fb7cd04fa0853d08459516a75d/src/slic3r/GUI/GLToolbar.cpp#L1383

buzzhuzz, 2024-03-20:
It may be hard to get sharp icons across the board since entire UI is rendered using OpenGL. OpenGL does not use integer coordinates and sizes and does not know about pixels on particular display.

I too would like to get crisp sharp text and icons across whole GUI, but as of now I'm not sure about level of effort required to do this.

PR #4552 (2024-05-18): Best effort for pixel perfect icon rendering
This currently requires the toolbar icon to have even size.

Fixes #4328

Before:
[image]

After:
[image]

Toolbar icons should be noticeably sharper.
closed 2024-05-18 by SoftFever