OrcaSlicer/OrcaSlicer #4337: Printer starts before reaching hotend temp - due to M104 command rather thanM109.  Previously in #2334 but still present

daveDGL, 2024-03-04:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

1.9.1

### Operating System (OS)

Windows

### OS Version

win11

### Printer

Voxlab aquila

### How to reproduce

its in the orca generated Gcode that comes before my start gcode:
M204 P10000 R1000 T10000
M205 X10.00 Y10.00 Z0.40 E5.00 ; sets the jerk limits, mm/sec
M190 S55 ; set bed temperature and wait for it to be reached
M104 S220 ; set nozzle temperature  <<<<<<<<<<<<<<
;TYPE:Custom

### Actual results

If I select upload and print and the printer is currently idle (cool hotend) the bed reaches temperature then as soon as the hotend temp starts to rise it starts the print - with nothing extruded for the first few seconds as the nozzle heats up.

### Expected results

The auto generated gcode should have M109 not M104 command for setting hotend temperature.  then the file progress would pause until this is reached.
[clip in TPU.zip](https://github.com/SoftFever/OrcaSlicer/files/14487201/clip.in.TPU.zip)

### Project file & Debug log uploads

[clip in TPU.zip](https://github.com/SoftFever/OrcaSlicer/files/14487214/clip.in.TPU.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Anything else?

this was previously reported as bug #2334 but was closed as stale. The comments /workarounds are all manual start code edits, which is not ideal given I print different materials so one setting would not work for all

mikigal, 2024-03-13:
I implemented a temporary fix that will work for all materials - the temperature is obtained from the first layer temperature. 
The code should be placed in your printer's Start G-Code section.

```gcode
M109 S{nozzle_temperature_initial_layer[initial_extruder]} ; Fix for printing without waiting for the temperature
```

daveDGL, 2024-03-13:
thanks - I discovered a similar fix after I posted the bug. I  am new to orcaslicer and didnt grasp that the the filament type had its own start gcode settings....I've  now added that M109 command for my 'go to ' filament in that area, but your solution is better as it works for all filament types/vendors and I've just added it.. I am really impressed with Orcaslicer.  Thanks
Dave

dragon-architect, 2024-06-08:
I'm +1'ing this bug with the Anycubic Kobra 2. I was literally in the process of starting my first serious test print and the hot end wasn't doing a preheat-and-wait. So I looked in the gcode output by Orca Slicer to find M104 to set the hot end temperature instead of M109. This seems like an oversight.

Visnitz, 2024-06-25:
I have a same issue with OS X Orca 2.1.0
Start G code normal. Filament G code usualy empty.
As I remember it used to work fine...

NathanNull, 2024-08-13:
Same problem, using Anycubic Vyper. Someone fix this pls.

Mirniy2000, 2024-08-16:
Правильно  понимаю что в кастомном  профиле также отсутствует данный М-код?

TrentBartlem, 2024-08-23:
Still there in 2.1.1

disconn3ct, 2024-11-22:
Did "orca bot" fix it? No? Then why is it being closed?

TrentBartlem, 2024-12-16:
The relevant code is [somewhere in here](https://github.com/SoftFever/OrcaSlicer/blob/main/src/libslic3r/GCodeWriter.cpp#L95-L112). 

Specifically, if you call this function with `wait=true` you'll probably get M109, otherwise you'll probably get M104.

I think the relevant usage of this is `_print_first_layer_extruder_temperatures`, and the relevant call is [here](https://github.com/SoftFever/OrcaSlicer/blob/main/src/libslic3r/GCode.cpp#L2362), based on the scenario that bed temperature is set to wait correctly, but not nozzle temp.

So maybe the solution is to set the last argument to `true` here?

torte71, 2024-12-28:
+1 for changing M104 to M109

After a M600 filament change, waiting for the extruder to reach target temperature is necessary as well.
One may get it around it by adding sth. like this to the "Filament start gcode" of **each** filament (or manually dialing in the new temp. when changing filament and wait for it to settle):
```
{if layer_num < 1}
M109 S{nozzle_temperature_initial_layer[current_extruder]}
{else}
M109 S{nozzle_temperature[current_extruder]}
{endif}
```

Using M104 (setting the temp. without waiting for it) is nice for heating up in the background, but it only makes sense, if the temp. is checked before actually start printing (e.g. if it is followed by a M109 later).
If Orca just used M109, it would still be possible to heat up the nozzle in the background while waiting for the bed to heat up in the "Machine start gcode" as stated above - it wouldn't cause an additional delay, as the nozzle is already heated up.
But setting the nozzle temp. without waiting for it before printing doesn't make any sense to me, or am I overlooking sth.?

EntwicklungsAnstaltGretsch, 2025-01-15:
#7582 is related

GreavesDesignerOne, 2025-03-15:
Same issue. Because the M109 was missing, the print started before the hotend could melt the filament. As a result, I'm one rather large step closer to having to disassemble and clean my extruder, which also means recalibrating a lot of stuff. Bummer!

RodentVienna, 2025-05-25:
also M600 user. Orca produces gcode with M104 instead M109 after Filament Change, which I believe is a mistake.

odddollar, 2025-06-18:
I've got the same issue with my custom profile for the Sovol SV01

RodentVienna, 2025-09-17:
unstale.

closed 2026-09-21 by SoftFever