OrcaSlicer/OrcaSlicer #4342: Add an option to disable the "Device" tab.

Ataraxia-Mechanica, 2024-03-05:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

The device tab in windows runs webview2 which is laggy and consumes a significant amount of CPU cycles even when in background. This results in basically the entire computer being unusable with 4+ Orca slicers open.

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

An option to disable the tab would be good. Or pause the tab while the page is in background.

Maximus-CZ, 2024-03-05:
Can not replicate, here the OrcaSlcier process consumes less than 1.5% CPU when on device tab. regardless if camera feed is running or not. When on other tabs the CPU usage falls to below 0.8%.
Windows 10, 7800x3d.

Ataraxia-Mechanica, 2024-03-05:
Could it have something to do with high resolution? I have a 4k monitor and each instance takes 20% processing power or so.

Maximus-CZ, 2024-03-05:
I too am running at 4k, altho at 1.25x scaling

On Tuesday, Mar 5, 2024, 12:06,  wrote:

> Could it have something to do with high resolution? I have a 4k monitor and each instance takes 20% processing power or so. You are receiving this because you commented.Message ID:

Ataraxia-Mechanica, 2024-03-05:
Guess this has something to do with my computer. Guess sending webview2 to the shadow realm via windows firewall it is then.

Ataraxia-Mechanica, 2024-03-05:
After blocking internet access Orca slicer only consumes ~0.2% CPU in background, despite all the webview2.exe running, so it's caused by the combination of fluidd + webview. Weird.

closed 2024-03-05 by Ataraxia-Mechanica