OrcaSlicer/OrcaSlicer #4347: "Sphere" isn't visible in paint or paint supports

ThinkBigPrintHuge, 2024-03-05:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

v2.0.0

### Operating System (OS)

Windows

### OS Version

Windows 11 Home

### Additional system information

CPU: i9-12900h
GPU: Nvidia 3070Ti (Laptop)
Ram: 32g LPDDR5

### Printer

n/a

### How to reproduce

1. Go to Paint/Support paint
2. Select "Sphere" tool
3. Hover over part

### Actual results

[image]
[image]
[image]
[image]

### Expected results

Sphere tool should be visible (as in previous versions).

### Project file & Debug log uploads

No need, this happens on all models since transitioning to V2.0.0 

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Anything else?

I searched for related issues and found a report about Sphere not showing up during Seam Painting, but no reference to painting and support painting. It's important to note I switched from 1.8.1 to 2.0.0, so I am unsure if this also happening in V1.9.x.

cobyn, 2024-03-13:
Can confirm this started happening on V1.9

foreachthing, 2024-03-13:
Still an issue in 2.0b.

closed 2024-06-20 by github-actions[bot]